bitcraze/crazyflie-firmware #937: Generic Bolt Discussion Issue

knmcguire, 2022-02-08:
With this PR #931 we added quite some persistent parameters but there are still some hardcoding we need to take care off.

1. #936 We need to decide what to do with the motor compensation function
2. Lighthouse different orientation (currently this will not be possible to implement)
3. Support for more than 1s battery and different battery thresholds
4. Different IMU orientation (not immediately handy for bolt but for different type of drones like Flapperdrones?)
5. Different motor type support (different platform)

@matejkarasek would it be possible to create 2 PRs for item 4 and 5, as these are quite handy for your platform?

matejkarasek, 2022-02-09:
Yes, I will create those but will in only in two weeks or so...

knmcguire, 2022-02-14:
no problem at all :) we are not in a hurry. Just let us know once it is online

knmcguire, 2022-03-07:
#974 is related to this