bitcraze/crazyflie-firmware #946: Kbuild: Add platform configs

jonasdn, 2022-02-17:
Add non-default configs to config/ folder.

Which should we have?

- [x] Bolt

- [ ] BQ

- [ ] Brushed Crazyflie?

- [ ] Flapper Drone?

knmcguire, 2022-03-07:
Perhaps a bit too early yet for the CF BL but perhaps we can already start for the Flapperdrones. Let us know what you think @matejkarasek

knmcguire, 2022-03-07:
https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/development/create_platform/

matejkarasek, 2022-03-08:
> Perhaps a bit too early yet for the CF BL but perhaps we can already start for the Flapperdrones. Let us know what you think @matejkarasek

Hey, yeah I was already briefly looking into that after your newsletter from 2 weeks ago... and am certain this is the way to go for Flapper!
It is on my to-do-list to try it out, might reach out then and we can create a PR together should I any assistance...

knmcguire, 2022-03-08:
We just had an internal discussion about this, and we do want to empathize that any platform config that we add, should be stable and not subjected to a lot of change. So that it should be possible to have base defines, and ofcourse any small variations can always be tuned and saved by persistent parameters. 

There has been some more discussions about this a while ago that I was not part of, so the idea is to write documentation about this policy for handling new platforms. I made a seperate issue on that.

matejkarasek, 2022-03-08:
Ok, understood... Well we can always maintain a platform config in our fork until things are stable enough, but this simplifies things already!

knmcguire, 2022-03-08:
Yeah that is indeed what I thought about that. We hope to write the policy done somewhere where we can also understand it and remember it as well.