bitcraze/crazyflie-firmware #949: Possible bug with i2cdevWrite - sending a single byte of data, sends two bytes

amitga, 2022-02-22:
Hi.
I recently ran into a problem while sending a single byte of data using `i2cdevWrite`:
My goal is to send/write a single byte of data to an I2C slave (no internal address/register).
When I send like this: `i2cdevWrite(I2C_DeckBus, devAddr, 1, &myByte);` actually my byte (uint8_t) value is sent as one byte, but there is also another empty (zero) byte sent immediately after that byte (within the same data transfer it seems).

I have confirmed this with various MCUs acting as an I2C slave (Arduino, PIC16f...).
This also happend if the data is a buffer/array of data, but we send `1` as the requested length.
If we choose to send more than `1` from a buffer, all works as expected.

I have looked into the code, and it seems related to the ISR handling the I2C events.
Specifically here:
https://github.com/bitcraze/crazyflie-firmware/blob/ef06e40eea76bf3aafb7389ef074c0af87054066/src/drivers/src/i2c_drv.c#L567

I think that the problem may be that the `messageIndex` is not initialised/reset when that single byte is sent.
I added `i2c->messageIndex = 0;` right after: https://github.com/bitcraze/crazyflie-firmware/blob/ef06e40eea76bf3aafb7389ef074c0af87054066/src/drivers/src/i2c_drv.c#L570
and it seems to solve the problem: working with sending a single and multiple bytes, as stated in data length when calling `i2cdevWrite` function.

But, I'm still not sure why this would happen in the first place, or if my quick fix is correct or can cause some other side-affects.

I would like to hear your thoughts about this.
BTW, I'm running a CF Bolt with a bit dated firmware, but its seems that the ISR logic hasn't changed much in `master` branch compared to mine (they are basically identical). I cannot confirm if this bug happens with newer/latest firmware (I would have to re-write my custom code).

Thanks!

knmcguire, 2022-03-02:
Hi Thanks for notifying us on this! It seem that this should not happen, but we will discuss it internally why this happens. Probably we can give you an update on that next week.

amitga, 2022-04-10:
Thanks.
Just an update: I ended up using `i2cdevReadReg8` instead, which seems to work correctly (doesn't send an additional byte).

whoenig, 2022-04-12:
Not sure if it's related, but https://github.com/bitcraze/crazyflie-firmware/tree/dev-systemId had some fixes(?) for I2C that we never ended up merging back to master, due to limited testing.

tobbeanton, 2022-06-09:
Fixing this with an added instruction is as good as it gets right now.

closed 2022-06-09 by tobbeanton