bitcraze/crazyflie-firmware #960: Kbuild: adding C++ support?

ntamas, 2022-02-23:
I'm testing the new Kbuild-based build system and trying to convert an app of our own to the new build system. One issue that I can't seem to work around is that some of the files in the app are written in C++ because of historical reasons. With the previous build system, I was able to add support for C++ files to the makefile simply by adding extra rules to the app-level Makefile like this:

```
CXX = $(CROSS_COMPILE)g++
LD = $(CROSS_COMPILE)g++

CXX_COMMAND=$(CXX) $(CFLAGS) -c $< -o $(BIN)/$@
CXX_COMMAND_SILENT="  CXX   $@"
.cpp.o:
     @$(if $(QUIET), ,echo $(CXX_COMMAND$(VERBOSE)) )
     @$(CXX_COMMAND)
```

It seems like this is not possible with the new Kbuild system because the app-level makefile eventually ends up invoking another `make` process with the top-level makefile, which does not have support for C++ files. In the previous build system, the top-level makefile was included instead of forking off to another `make` process so it was easy to add the modifications.

Would it be possible to add support for C++ source files directly in `tools/make/targets.mk` even if the current Crazyflie firmware does not use any C++ sources?

jonasdn, 2022-02-23:
Hi @ntamas !

Could you try the linked pull-request and see if it fixes stuff for you?

Thanks!

ntamas, 2022-02-23:
It seems like it's going to be a bit more complicated. In order for the linking stage to work properly, one probably needs to use `$(CROSS_COMPILE)g++` as the linker and not `$(CROSS_COMPILE)gcc` if at least one C++ file is involved in the build process. I'm not sure whether there's an easy way around it to handle this in `kbuild` itself; right now I'm just patching `tools/kbuild/Makefile.kbuild` to use the appropriate linker.

jonasdn, 2022-02-24:
> It seems like it's going to be a bit more complicated. In order for the linking stage to work properly, one probably needs to use `$(CROSS_COMPILE)g++` as the linker and not `$(CROSS_COMPILE)gcc` if [...]

Ok. What issues are you seeing in linking and what is your current work-around? Let's see if we can manage an ok permanent solution.

jonasdn, 2022-02-24:
Does it work for you if you keep linking with `gcc` but add `-lstdc++` when C++ files are included?

ntamas, 2022-02-24:
`-lstdc++` gets rid of a few of the undefined references but not all of them. I could not figure out how to add `-lstdc++` in an app-level `Kbuild` file (`libs-y += -lstdc++` seemed to have no effect) so I just added it to `../tools/kbuild/Makefile.kbuild` straight before `-lm` and then it was picked up properly.

The next problem was a missing reference to `abort()`. `abort()` is called implicitly by the `new` operator in C++ in case the allocation fails. I solved it by adding `obj-y += abort.o` to `src/utils/src/Kbuild`.

Next up: undefined references to `_write`, `_close`, `_fstat` and many others, none of which were called directly by our C++ code. It took me a while to figure out that `assert()` is calling these implicitly, so if you have an assertion anywhere in your C++ code, that's a problem. I solved it by adding `subdir-ccflags-y += -DNDEBUG` to our app-level `Kbuild` file to force assertions to be removed. I have no idea why this worked in the pre-Kconfig days.

And now I'm left with a single undefined reference:

```
/opt/homebrew/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /opt/homebrew/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc.a(lib_a-sbrkr.o): in function `_sbrk_r':
sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk'
collect2: error: ld returned 1 exit status
```

It turns out that `sbrk` is a low-level function that is required by `malloc` / `free` / `new` / `delete`, so this is not really related to C++ any more but it's more about dynamic memory allocation by code that is not aware of FreeRTOS and its `pvPortMalloc()` and `vPortFree()` calls. I can work around this by providing a `malloc` / `free` wrapper around the FreeRTOS functions, but I can't figure out why it worked with the pre-Kconfig version.

So, this is the full patch so far (without the `-DNDEBUG` flag because I added that in the app layer):

```patch
diff --git a/src/utils/src/Kbuild b/src/utils/src/Kbuild
index 69a6100b..1b8a5146 100644
--- a/src/utils/src/Kbuild
+++ b/src/utils/src/Kbuild
@@ -5,6 +5,8 @@ obj-y += cpuid.o
 obj-y += crc32.o
 obj-y += debug.o
 obj-y += eprintf.o
+obj-y += abort.o
+obj-y += malloc.o

 ### Add rules for handling generated version.c

diff --git a/src/utils/src/malloc.c b/src/utils/src/malloc.c
new file mode 100644
index 00000000..48cfa7a4
--- /dev/null
+++ b/src/utils/src/malloc.c
@@ -0,0 +1,22 @@
+#include "FreeRTOS.h"
+
+void* malloc(size_t size)
+{
+    void* ptr = NULL;
+
+    if (size > 0)
+    {
+        ptr = pvPortMalloc(size);
+    }
+
+    return ptr;
+}
+
+void free(void* ptr)
+{
+    if (ptr)
+    {
+        vPortFree(ptr);
+    }
+}
+
diff --git a/tools/kbuild/Makefile.kbuild b/tools/kbuild/Makefile.kbuild
index ee2bcc70..92a30fae 100644
--- a/tools/kbuild/Makefile.kbuild
+++ b/tools/kbuild/Makefile.kbuild
@@ -553,7 +553,7 @@ firmware-libs       := $(patsubst %,%/lib.a, $(libs-y))
 firmware-all   := $(firmware-objs) $(firmware-libs)

 quiet_cmd_firmware = LD      $@
-      cmd_firmware = $(LD) $(image_LDFLAGS) $(LDFLAGS) $(firmware-objs) $(firmware-libs) -lm -o $@.elf
+      cmd_firmware = $(LD) $(image_LDFLAGS) $(LDFLAGS) $(firmware-objs) $(firmware-libs) -lstdc++ -lm -o $@.elf

 firmware: $(firmware-all) FORCE
        +$(call if_changed,firmware)
```

I haven't tried uploading the compiled firmware to a drone yet; that's for tomorrow. (Well, today, but after the next sleeping cycle).

jonasdn, 2022-02-25:
So, I am trying to reproduce this! I would like your help.

I tried to create a more complex example C++ application, and then try to build and link it with C++. I could not get i to fail :)

My branch is here: https://github.com/bitcraze/crazyflie-firmware/tree/jonasdn/g%2B%2B-link-experiment

And contains one commit:

```
From b1dd31f4fea1d789721a0bb3aea76f184b91acc5 Mon Sep 17 00:00:00 2001
From: Jonas Danielsson <jonas@bitcraze.io>
Date: Fri, 25 Feb 2022 09:40:38 +0100
Subject: [PATCH] oot.mk: Add OOT_USES_CXX

Create a more complex example cpp app and create a way to force
linking with g++
---
 .../app_hello_world-cpp/src/hello_world.cpp    | 18 +++++++++++++++++-
 tools/kbuild/Makefile.kbuild                   |  8 +++++++-
 tools/make/oot.mk                              |  4 ++++
 3 files changed, 28 insertions(+), 2 deletions(-)

diff --git a/examples/app_hello_world-cpp/src/hello_world.cpp b/examples/app_hello_world-cpp/src/hello_world.cpp
index ce1ef99a..1b1c9fa9 100644
--- a/examples/app_hello_world-cpp/src/hello_world.cpp
+++ b/examples/app_hello_world-cpp/src/hello_world.cpp
@@ -7,7 +7,7 @@
  *
  * Crazyflie control firmware
  *
- * Copyright (C) 2019 Bitcraze AB
+ * Copyright (C) 2022 Bitcraze AB
  *
  * This program is free software: you can redistribute it and/or modify
  * it under the terms of the GNU General Public License as published by
@@ -31,6 +31,9 @@
 #include <stdint.h>
 #include <stdbool.h>
 
+#include <cassert>
+#include <string>
+
 extern "C"
 {
   #include "app.h"
@@ -43,6 +46,13 @@ extern "C"
 
 #define DEBUG_MODULE "HELLOWORLD"
 
+
+class MyClass {
+  public:
+    int myNum;
+    std::string myString;
+};
+
 void appMain()
 {
   DEBUG_PRINT("Waiting for activation ...\n");
@@ -51,4 +61,10 @@ void appMain()
     vTaskDelay(M2T(2000));
     DEBUG_PRINT("Hello World!\n");
   }
+
+  MyClass *cl = new MyClass();
+
+  DEBUG_PRINT("MyClass has a num: %d\n", cl->myNum);
+
+  assert(1);
 }
diff --git a/tools/kbuild/Makefile.kbuild b/tools/kbuild/Makefile.kbuild
index 06536b27..71c82cfd 100644
--- a/tools/kbuild/Makefile.kbuild
+++ b/tools/kbuild/Makefile.kbuild
@@ -245,7 +245,13 @@ include scripts/Kbuild.include
 
 # Make variables (CC, etc...)
 AS		= $(CROSS_COMPILE)as
-LD		= $(CROSS_COMPILE)gcc
+
+ifndef ($(OOT_USES_CXX),)
+LD      = $(CROSS_COMPILE)g++
+else
+LD      = $(CROSS_COMPILE)gcc
+endif
+
 CC		= $(CROSS_COMPILE)gcc
 CXX		= $(CROSS_COMPILE)g++
 CPP		= $(CC) -E
diff --git a/tools/make/oot.mk b/tools/make/oot.mk
index a5206e1d..8464843c 100644
--- a/tools/make/oot.mk
+++ b/tools/make/oot.mk
@@ -12,6 +12,10 @@ OOT_CONFIG ?= $(OOT)/oot-config
 
 OOT_ARGS ?= -C $(CRAZYFLIE_BASE) OOT=$(OOT) EXTRA_CFLAGS=$(EXTRA_CFLAGS)
 
+ifneq ($(OOT_USES_CXX),)
+OOT_ARGS += OOT_USES_CXX=1
+endif
+
 .PHONY: all clean
 
 all:
-- 
2.33.1

```

This compiles when I go:

```
$ cd examples/app_hello_world-cpp
$ rm -rf build
$ make -j 12 V=1
```

How do I make it fail? Or does this `OOT_USES_CXX` trick work for you?

jonasdn, 2022-02-25:
ping @ataffanel

ntamas, 2022-02-25:
I'm in a meeting until the afternoon but I'll try to break it later :) It also works for me as is, so I need to figure out what's the difference between our "real" codebase and this example.

jonasdn, 2022-02-25:
Could it be the `extern "C" {` to force C linkage? You might need to sprinkle it around where the C++ functions are used?

ntamas, 2022-02-25:
Okay, so, first of all, you need `ifneq ($(OOT_USES_CXX),)` in `tools/kbuild/Makefile.kbuild`, not `ifndef`.

Then, the compiler is smart and it knows that the execution can never reach the part after the `while (1)` infinite loop, so anything you put in there is ignored. Moving the construction of `MyClass` above the while loop yields this:

```
/opt/homebrew/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: [...]/examples/app_hello_world-cpp/built-in.o: in function `appMain':
[...]/examples/app_hello_world-cpp/src/hello_world.cpp:61: undefined reference to `operator new(unsigned int)'
collect2: error: ld returned 1 exit status
```

Changing to the C++ linker still yields the same error, maybe because of `-nostdlib`? I'm not sure but I then added linking to `stdc++` explicitly (by placing it in front of `-lm`), which gives me this:

```
/opt/homebrew/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /opt/homebrew/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc.a(lib_a-abort.o): in function `abort':
abort.c:(.text.abort+0xa): undefined reference to `_exit'
/opt/homebrew/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /opt/homebrew/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc.a(lib_a-sbrkr.o): in function `_sbrk_r':
sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk'
/opt/homebrew/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /opt/homebrew/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc.a(lib_a-signalr.o): in function `_kill_r':
signalr.c:(.text._kill_r+0x12): undefined reference to `_kill'
/opt/homebrew/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /opt/homebrew/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc.a(lib_a-signalr.o): in function `_getpid_r':
signalr.c:(.text._getpid_r+0x0): undefined reference to `_getpid'
```

You can see that the error about `operator new` is now gone; the ones above are probably due to the usage of `assert()`, and note that `_sbrk` also appeared, so there you go, I think this pretty much reproduces what I've seen in our more complex setup.

knmcguire, 2022-03-08:
After internal discussion, we are fine of having the cpp example as long as it works with building it and it doesn't break in the CI. We see this as a valuable contribution but currently we don't have the resources right now to fix the remaining issues with the CPP app. 

Currently the CPP version hello_world seems to be running fine in our CI and locally, but if there is any issues remaining on your side still, we would advise to try to fix it on your end en perhaps make an PR on our side if there is anything that can be fixed in the firmware. 

Would that be okay for you? We really need to prioritize our tasks now as we are with a smaller number of people..

ntamas, 2022-03-08:
The `hello_world_cpp` example runs fine on CI only because the compiler is smart enough to figure out that anything after the infinite loop in `appMain()` will not get executed ever, so it's not generating any code for the part after the infinite loop -- therefore no calls to the `new` operator and no calls to `abort()` either. Moving the `new` call to above the main loop triggers the error on my end and would probably also trigger the error in CI.

Anyhow, I'll prepare a PR in the next few days to help move this forward.

knmcguire, 2022-03-08:
that would be awesome, thanks a lot!

PR #962 (2022-02-23): Support C++ in app-layer
- Kbuild: Add support for C++ (.cc and .cpp) source
- app-layer: Add C++ example app

Fixes: #960
PR #985 (2022-03-09): Add support for new(), malloc(), free() and assert() in C++ app code
This PR attempts to fix #960 for more complex C++ apps that uses classes or assertions. See the discussion in #960 for more details.

The PR extends the build system to use the C++ linker and to link to `stdc++` when the OOT app makefile indicates that it needs C++ facilities by setting `OOT_USES_CXX` to a non-empty value. The PR also provides implementations of `malloc()` and `free()` in terms of FreeRTOS functions to ensure that C++ code that was not designed with FreeRTOS in mind keeps on working even in this environment.
closed 2022-03-09 by krichardsson