bitcraze/crazyflie-firmware #978: Ready To Fly Mode

Moj-Dev, 2022-03-04:
I'd like to define a ready to fly mode in my application that motors spins in lowest rpm before takeoff. Could you give me help that what is the best way to implement this?

Moj-Dev, 2022-03-04:
I implemented it using python library. Here is the code you can use:
        roll = 0
        pitch = 0
        yawrate = 0
        print("let's go")
        thrust_mult = 1
        thrust_step = 500
        thrust = 20000
        cf.commander.send_setpoint(0, 0, 0, 0)
        for r in range(10):
            cf.commander.send_setpoint(roll, pitch, yawrate, thrust)
            time.sleep(0.1)
        print("Ready 2 fly")

knmcguire, 2022-03-07:
Hi! I see that you asked about this in the forum as well: https://forum.bitcraze.io/viewtopic.php?t=5268 . If you could try to prevent double posting to keep it a little manageable for us?

Usually these type of support questions we usually prefer to answer on the forum, but it is an interesting discussion that I can link to another issue that is a bit older, namely #292. We noticed that the take off overshoot was significantly less if the motors were already spinning before take off.

I guess then the discussion would be, is this a better functionality to have in the [python lib](https://github.com/bitcraze/crazyflie-lib-python), or directly in the firmware?

tobbeanton, 2022-03-08:
This functionality is already present but still a bit under development and yet no documentation. There is the parameter powerDist.idleThrust to which the thrust levels to the motors will be set when the system is armed. For the CF2.1, the system is armed from start but this can be changed so that the parameter system.forceArm has to be set to 1 to enable the motors. If you are using firmware with Kbuild, under "Motor configuration" enable "Set disarmed state after boot". If still using config.mk add CFLAGS += -DSTART_DISARMED.

closed 2022-03-09 by knmcguire