bitcraze/crazyflie-firmware #980: Remove crtpCommanderHighLevelStop

knmcguire, 2022-03-07:
Since PR #903 has been merged and the discussion coming to a close at #861, there is still one thingy remaining to really close this off, which is to remove the crtpCommanderHighLevelStop.

It is not blocking anything and it is not a bug, but it just has been deprecated. I guess we should also talk about how to properly do that for crtp functionality.

ntamas, 2022-03-07:
If you do decide to deprecate it, please make it clear what's the preferred alternative in third-party app code. Currently we use it [like this](https://github.com/skybrush-io/crazyflie-firmware/blob/cm-develop/app/src/drone_show.c#L642) in an app.

knmcguire, 2022-03-07:
Good point. perhaps we have come to a point that we can't just remove it like that... but the code will not stay clean either.

Btw, we discussed this dillemma in this issue too: #849,

knmcguire, 2022-03-08:
I've made an discussion issue on this #986 on the matter of deprecating functionality.

knmcguire, 2023-02-15:
Just a mention that we will need to say in the parameter documentation that this highlevel commander parameter is deprecated and doesn't do anything functionally anymore:
https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/api/params/#commander

ntamas, 2023-02-15:
Please also mention alternatives for people who are migrating old code away from `crtpCommanderHighLevelStop()` (like yours truly). Shall I just set `commander.enHighLevel` to zero?

knmcguire, 2023-02-16:
So currently the parameter commander.enHighLevel does nothing. Before, it needed to be explicitly mentioned that somebody was about to send high-level commands, which this parameter was used, but now the switch between high level and low level is done automatically due to this PR https://github.com/bitcraze/crazyflie-firmware/pull/903. In that sense, there is no alternative because it is automatic. 

About crtpCommanderHighLevelStop() is something different as that is actually still functional. It stops the planner of the high-level commander, therefore not generating any further setpoints. An alternative functionality can be achieved with a combination of `commanderNotifySetpointsStop` with sending another setpoint right away: https://github.com/bitcraze/crazyflie-firmware/blob/56e357e0f4a3f01e04b0849214ef3628292f76d2/src/modules/interface/commander.h#L49-L54

But, how things are, is that we don't have the exact same functionality replacement for it, and according to the new API deprecating policy we just set up (see this PR https://github.com/bitcraze/bitcraze-website/pull/152), we will once deprecate indicate the alternative. But for now it is fine :) just keep an eye on the release notes once we decide to deprecate it once we have figured out the right alternative that is more generic for not only the high level commander.

knmcguire, 2023-03-20:
Now we actually have a [deprecation policy](https://www.bitcraze.io/development/development-overview/#apis-versioning-and-evolution-of-the-code-base)! commander.enhighlevel will disappear after august 2023 and there is no need for an alternative as the high level commander already automatically switch to highlevel commander as soon as you send a high level command. So anybody that is using this in their script can remove this parameterset to prevent errors in future releases when it is removed.

closed 2023-03-20 by knmcguire