bitcraze/crazyflie-firmware #981: Make initial yaw variance in Kalman filter configurable

ntamas, 2022-03-07:
It looks like the Kalman core assigns a [fairly low variance](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/kalman_core/kalman_core.c#L132) to the initial yaw. In our experience it can be a cause of unexpected flips after takeoff when using the Loco positioning system and the drone is not placed facing the X axis very accurately. We raised this to 0.2 (i.e. 11.5 degrees) in our fork to reduce the chance of flips, and it did not seem to have a negative effect on overall flight performance so I was wondering whether it would be possible to make this option configurable, either with a persistent parameter or a firmware configuration switch.

knmcguire, 2022-03-07:
Ah yes great find! And yes I would agree that this should be configurable and the variance should probably be higher. To take it further, probably all those std's on those page should be configurable...

The thing is that the gyro seems to be integrated in the covariance update of the EKF, instead of added as a measurement with a standard deviation like in regular kalman filters... so I'm not sure what the full effect would be. 

Probably it would be best to just go ahead and testt this for three cases:
- EKF with manual flight
- EKF with Loco or flow deck (no extra yaw measurements)
- EKF with lighthouse or mocap (with extra yaw measurement)

knmcguire, 2022-03-08:
It is related to this ticket I opened, to somewhere document the policy #948 

But to come back to this discussion, this is an initial standard deviation that shouldn't change during runtime. I would make a strong case that this should be a define, and configurable in kbuild.

jpreiss, 2022-03-22:
Tangentially, it would be nice if controllers could be aware of the current EKF belief variance for yaw and use it to modify their gain. Controllers should not be trying to correct yaw error with high gain unless the variance is small.

gemenerik, 2025-03-18:
I believe the observation in the original post makes logical sense. I am working on yaw estimation in the Kalman filter for the Lighthouse system and have noticed that the default initial standard deviation causes very slow convergence to any initial yaw other than zero. While the Loco Positioning System (LPS) differs from Lighthouse in that yaw is not directly observable, movement in the x-y plane makes that orientation observable again for the Kalman filter (similar for other orientation axes). Colleagues have described damped spiraling behavior during which the Kalman filter gradually corrects the yaw estimate error, demonstrating this. Increasing the initial standard deviation for yaw should allow the filter to correct itself more quickly.

As part of our work on the Lighthouse system, we plan to automatically adjust the initial standard deviation based on the hardware in use. We can use the same implementation here, but the actual value will be different because the required initial standard deviation for Lighthouse will likely be much higher. In Lighthouse, yaw can be initialized 180 degrees off, whereas Loco users are expected to set it as close to zero as possible.

Regarding configurability, having a persistent parameter for these Kalman filter settings could be useful for tuning, as it would allow us to test different values without recompiling. However, once we settle on an optimal solution for the main firmware, this configurability may become less relevant. I’m not sure if there are general guidelines for deciding what should become a persistent parameter, but it's likely that once we establish a good default, it won’t need to be adjusted further.

As for controller awareness of current variance, @jpreiss, if this is still of interest, we should open a separate issue to keep this one focused and easily resolvable.