bitcraze/crazyflie-firmware #982: make cload wsl broken

knmcguire, 2022-03-07:
With the latest changes, `make cload` seems to be broken for WSL:

```
tools/kbuild/Makefile.kbuild:147: warning: overriding recipe for target 'cload'
Makefile:136: warning: ignoring old recipe for target 'cload'
make[1]: Entering directory '/home/kimberly/Development/c/crazyflie-firmware/build'
  GEN     ./Makefile
scripts/kconfig/conf  --silentoldconfig Kconfig
python3 -m cfloader  flash  cf2.bin stm32-fw
Restart the Crazyflie you want to bootload in the next
 10 seconds ...
Failed to flash: Cannot find a Crazyradio Dongle
make[1]: Leaving directory '/home/kimberly/Development/c/crazyflie-firmware/build'
```

it should write python.exe. The warning indicates that the recipe for cload is overwritten, but not exactly sure how this happens from the kbuild file.

knmcguire, 2022-03-08:
This happens in Ubuntu too btw, ofcourse it does not have as drastic effect, but it is still bad that this is overwritten... 

It has something to do with[ the line](https://github.com/bitcraze/crazyflie-firmware/blob/acd7275370b153a0e6aa7840f12dab344b7d3315/tools/kbuild/Makefile.kbuild#L147) mentioned in the Makefile.kbuild file in tools

```
$(filter-out _all sub-make $(CURDIR)/Makefile, $(MAKECMDGOALS)) _all: sub-make
	@:
```

krichardsson, 2022-03-09:
As far as I understand the root problem for this issue was actually that `make cload` does not work on Windows as make tries to use the wrong python binary (python3 instead of python.exe), correct @knmcguire ?
The pull request #989 solves that problem

The warning 

    tools/kbuild/Makefile.kbuild:147: warning: overriding recipe for target 'cload'
    Makefile:136: warning: ignoring old recipe for target 'cload'

does not seem to be related to this problem, I do not know why we get that and it still remains to be solved.

PR #989 (2022-03-10): Remove re-definition of PYTHON
PYTHON is redefined in Makefile.kbuild which over writes the platform specific definition in the root make file.

Closes #982
closed 2022-03-10 by knmcguire