bitcraze/crazyflie-firmware #1017: keep tilt compensation?

knmcguire, 2022-04-13:
In efforts to make a python binding of the controllers, we have come across a potential problem. We noticed that the PID controller is a bit too coupled with the state estimation in the sense that it is relying on some header files and functions. Most header files could be already be removed with more generic helper header files, but the #include "sensfusion6.h" of [controller_pid.c](https://github.com/bitcraze/crazyflie-firmware/blob/493588141809e769ff9224672535fcdb4ce92bb3/src/modules/src/controller_pid.c#L6) is in particular challenging. This gives a direct link with the complementary filter, and ideally we would really to decouple controllers and state estimators.

So technically, the boolean [tiltCompensationEnabled](https://github.com/bitcraze/crazyflie-firmware/blob/493588141809e769ff9224672535fcdb4ce92bb3/src/modules/src/controller_pid.c#L16) is always enabled on false ever since the kalman filter has been introduced about 6 years ago by @mikehamer, but there must be a reason why it was implemented in the first place?

Questions:
- Why was this tilt compensation  in the first place?
- Is anybody still using it after 6 years, ever since the default is false?
- Will this still be useful in the future, for let's say manual control of bigger platforms?

tobbeanton, 2022-04-13:
The tilt compensation is a manual flying aid to make it a bit easier to control the thrust while doing roll or pitch. I don't think anyone uses it and would say it's pretty safe to remove it.

It might be useful in the future but then let's implement it in another, better way.

knmcguire, 2022-04-19:
Just reopening this as this can still be an added enhancement in the future.