bitcraze/crazyflie-firmware #1032: Emergency stop does not stop the motors if idleThrust is set

tobbeanton, 2022-04-22:
How to reproduce:
1. Connect the a crazyflie using the cfclient
2. Set parameter powerDist.idleThrust to e.g. 7000
3. Turn the crazyflie up side down and the motors doesn't stop

victorhook, 2022-11-08:
Tried this with firmware 2022-09. Failed to reproduce the issue, seems solved.

matejkarasek, 2022-11-08:
I just tested this with CF2.1 on 2022-09 and indeed the motors do stop.

However, I also tested the Flapper platform on master (Bolt 1.1 & brushless motors) and there the motors keep running. So could be affecting only brushless motors...

@tobbeanton was it a brushless crazyflie where this was observed?

matejkarasek, 2022-11-10:
So please please disregard my previous comment, all works fine also on the Bolt-powered flapper platform with brushless motors, i.e. the motors in idle do stop upon flip as expected

The flapper platform has tumble check disabled by default... :)

knmcguire, 2023-05-02:
Maybe this will be solved with the supervisor work: https://github.com/bitcraze/crazyflie-firmware/pull/1272 ?

ArisMorgens, 2025-10-31:
This has been resolved! Closing.

closed 2025-10-31 by ArisMorgens