bitcraze/crazyflie-firmware #1033: How to limit max thrust for brushless / BigQuad build?

schlagenhauf, 2022-04-25:
Based on the BigQuad deck I have assembled a brushless quadcopter. It seems to work fine so far but I am hesitant to mount props because at full thrust the quad will be quite violent.

I'd like to reduce the maximum thrust / scale down the thrust mapping, but I'm unsure where. Reading through `motor.c` and `stabilizer.c` didn't yield any clues. Ideally the new limits would also be visible to the stabilizer to not create weird control behavior.

It seems #365 is related?

Thanks!

tobbeanton, 2022-04-29:
This is related to #365 and still not implemented. You could do something temporary meanwhile but since I have not looked into it in detail I'm not 100% sure what the best solution is. Ultimately you would like to limit it at the driver level so it would be "impossible" to go over the limit. Looking at it quickly that is a bit complicated and I instead suggest to implement another limit function in [power_distribution_stock.c](https://github.com/bitcraze/crazyflie-firmware/blob/2a282b575c2541d3f3b4552296952a6cc40bf5b5/src/modules/src/power_distribution_stock.c#L54).

schlagenhauf, 2022-05-09:
Thanks for your help!

Based on your hints i made this hack that seems to work (in that it limits the motor speed, no flight tests yet): https://github.com/schlagenhauf/crazyflie-firmware/tree/thrust_limit_hack.

Definitely not a proper solution, but i can keep my fingers. :partying_face: