bitcraze/crazyflie-firmware #1036: AI-deck UART communication fails

krichardsson, 2022-05-03:
We have seen that in some cases the communication with the ESP on the AI-deck fails. The communication is done over UART2 and problem is detected when checking the CRC on data received from the ESP 

https://github.com/bitcraze/crazyflie-firmware/blob/e6af56838b3a48f2da6f0fc31a4d808bada7cd11/src/deck/drivers/src/aideck.c#L196

It seems to happen when booting some times

tobbeanton, 2022-05-06:
I stated an investigation and can not easily reproduce it. I increased the uart speed to 1Mbit and measured the time in the USART2_IRQHandler interrupt routine. Receiving a 8bit takes ~2us - 4us and at 1Mbit there is a new potential 8bit every 8us. Suggestion is that we should rewrite the routine to assemble the packet in the ISR as the xQueueSendFromISR takes most of the time. Similar to what uart_syslink does. DMA might be the optimal solution but would conflict with DSHOT if this is used.

krichardsson, 2022-05-10:
I tried to use a StreamBuffer instead but the time spent in the interrupt is similar. The benefit would be that the waiting task could request chunks of bytes and would only be notified one time when the chunk is received, as opposed to the current solution where it is notified for every byte. We don't really know that it is running though, it is possible that data is pushed to the queue and handled in batches anyway. It is unclear what the impact on a system level is.

PR #1044 (2022-05-10): Update UART2 to work better with the AI-deck
We decided to NOT make any major changes to the UART implementations for now, instead we increased the buffer size for incoming data to be large enough for a full CPX packet.

Also added missing yields in interrupt handlers and fixed build problems when the UART2_LINK_COMM flag is enabled. 

(hopefully) Fixes #1036
closed 2022-05-10 by krichardsson