bitcraze/crazyflie-firmware #1073: paramSetFloat() seems to be broken for persistent parameters

ntamas, 2022-07-03:
It seems like paramSetFloat() does not work when it is invoked for persistent parameters because the `(params[varid.index].type & (~PARAM_CORE)) == PARAM_FLOAT` condition does not hold if the `PARAM_EXTENDED` bit is set.

I worked around it with the following patch, but it's unclear to me whether this is the right thing to do as this would happily attempt to set an int parameter with a `float` value bit-by-bit; on the other hand, `paramSetInt()` does the same thing the opposite way (allowing the user to set a float parameter with an int value, again bit-by-bit). Furthermore, both variants allow the user to overwrite parameters marked as read only.

```patch
--- a/src/modules/src/param_logic.c
+++ b/src/modules/src/param_logic.c
@@ -581,14 +581,9 @@ void paramSetFloat(paramVarId_t varid, float valuef)
   pk.data[0] = MISC_VALUE_UPDATED;
   pk.data[1] = varid.id & 0xffu;
   pk.data[2] = (varid.id >> 8) & 0xffu;
-  pk.size=3;
-
-  if ((params[varid.index].type & (~PARAM_CORE)) == PARAM_FLOAT) {
-      *(float *)params[varid.index].address = valuef;
+  pk.size = 3;

-      memcpy(&pk.data[2], &valuef, 4);
-      pk.size += 4;
-  }
+  pk.size += paramSet(varid.index, (void *)&valuef);
```

Ultimately, it should be decided where the type and readonly checks belong; if they do not belong to `paramSetInt()` and `paramSetFloat()`, then the patch above would work, but it then allows the user to do all sorts of nasty things to parameters from within an app. If they do belong there, then the checks should be added so my patch is not adequate.

knmcguire, 2022-07-04:
thanks @ntamas! 

@krichardsson has been working on this lately but as he is on his holiday I'll have @tobbeanton look this

ntamas, 2022-07-05:
FYI, it looks like `paramGetFloat()` is also broken; it mistakenly treats the parameter as integer when the `PARAM_EXTENDED` bit is set. A quick fix for the problem is this patch:

```patch
--- a/src/modules/src/param_logic.c
+++ b/src/modules/src/param_logic.c
@@ -543,7 +543,7 @@ float paramGetFloat(paramVarId_t varid)
 {
   ASSERT(PARAM_VARID_IS_VALID(varid));

-  if ((params[varid.index].type & (~(PARAM_CORE | PARAM_RONLY))) == PARAM_FLOAT)
+  if ((params[varid.index].type & (~(PARAM_CORE | PARAM_RONLY | PARAM_EXTENDED))) == PARAM_FLOAT)
     return *(float *)params[varid.index].address;

   return (float)paramGetInt(varid);
```

although at this point it would probably be more readable to use `PARAM_TYPE_MASK`.

tobbeanton, 2022-07-06:
Summary, the internal param API has not been tested...

I guess the intention with splitting into Integer and Float functions is to provide a bit of safety, but I don't know the original thinking. I suggest that we fix the functionality and keep a bit of the safety. I can give it a shot.

closed 2022-07-08 by tobbeanton

ntamas, 2022-07-08:
Quick feedback: I tested #1077 and it works great for our use-case, thanks!

knmcguire, 2022-07-11:
Tobias is on a holiday now but good to know this fixed your issue!

PR #1077 (2022-07-08): Improved internal parameter set/get functions
As a first step this PR will fix #1073 but it will need more work as the functions to internally set/get params need a clearer definition and error handling.

The checks to evaluate the type of param has been moved to an assert and made simpler. For the future I'm not so sure asserts are such a good idea but this would have to be part of a bigger redesign of the firmware. I am now only checking the type (PARAM_TYPE_INT and PARAM_TYPE_FLOAT) and if it is setting a param with PARAM_RONLY.