bitcraze/crazyflie-firmware #1086: Uart1 bad receiving with latest crazyflie-firmware

juliencasanovas, 2022-07-21:
Hi, 

I'm currently facing issues while receiving data on uart1 device into dedicated task from my custom app layer.
My main goal is to receive ASCII data from GAP8 device to crazyflie stm32 device through uart1 at 9600 bps.
I've run the code below onto two different releases of the crazylie firmware. 
- tag 2021.06 -> **working** 
- main : latest commit 46f497a1ef058750f59277c88578f9f2c13ea670 -> **not working**
I'm unable to decode correctly data into the latest firmware update. byte read is equal to -1 instead of actual ASCII value

uart is initialized as this : 
```
void appMain()
{
	uart1Init(9600);
```

And receiving task as this : 
```
xTaskCreate(UART_RX_Task,				// pointer to the task entry function
					      UART_RX_TASK_NAME,		// name for the task
					      1024,	// number of words for task's stack
					      NULL,						// the task's parameter
					      10,		// task priority
					      NULL);	
```

Function executed in task UART_RX_Task is like this : 

```
static bool getSteeringAngleGAP(float *steeringAngle)
{
  uint8_t data;
  bool getDataRX = 1;
  *steeringAngle = 0.0f;

  memset(dataRX, 0, sizeof(dataRX));
  int i = 0;
  DEBUG_PRINT("Waiting for header\n");

  while(getDataRX)
  {
	  getDataRX = uart1GetDataWithDefaultTimeout(&data);
	  if(data == '$')
	  {
		  DEBUG_PRINT("start of command\n");
		  getDataRX = 0;
		  return true;
	  }
  }

  return false;
```

knmcguire, 2022-07-21:
There have been quite some changes to the uart lately, mostly due to this PR: #1068

Could you revert the firmware code back to one week ago: b45f54fad1ee968e21b0f992bf626539a589edf5 and see if this still works?

You should probably also check the changes done in that PR to check how it affects your application in terms with UART1

knmcguire, 2022-09-06:
We will have a new release very soon. So please retest after that after updating and reopen this issue if it still a problem

closed 2022-09-06 by knmcguire