bitcraze/crazyflie-firmware #1111: Crash when overlapping commands in high level commander

krichardsson, 2022-09-07:
### Discussed in https://github.com/orgs/bitcraze/discussions/260

<div type='discussions-op-text'>

<sup>Originally posted by **ItsVipra** September  6, 2022</sup>
Hello Bitcraze community,

We've extensively been using a modified version of the synchronisedSequence.py script to fly with our swarm.
During our testing, we've discovered an issue using the high level commander, when overlapping commands are sent to the drone.

When the drone is sent a sufficiently long goto command (>~3s), while another one is currently being executed, the drone will fly along the inverse vector of the first command indefinitely (or at least long enough to crash in all our testing).

For example:

If the first command moves towards +x and -z the drone will move towards -x and +z once interrupted.
From logging data we know the drone's positioning data is unaffected, but the commander returns a setpoint ~20 m away from the current position, instead of the coordinates specified in the second command.
[image]

Of course an easy fix is to just not send overlapping commands, however this behavior seems unintended.
Below is a slightly modified synchronisedSequence.py you can use to recreate this behavior. Test at your own risk!

```
'''
this synchronized sequence file is unmodified except for the STEP_TIME constant, uris and sequence

the sequence consists of a takeoff, 
a short goto command, 
a long goto command, which is triggered before the first command finishes
and a land command for safety
'''
import threading
import time
from collections import namedtuple
from queue import Queue

import cflib.crtp
from cflib.crazyflie.swarm import CachedCfFactory
from cflib.crazyflie.swarm import Swarm

# Time for one step in second
STEP_TIME = 0.1

# Possible commands, all times are in seconds
Takeoff = namedtuple('Takeoff', ['height', 'time'])
Land = namedtuple('Land', ['time'])
Goto = namedtuple('Goto', ['x', 'y', 'z', 'time'])
# RGB [0-255], Intensity [0.0-1.0]
Ring = namedtuple('Ring', ['r', 'g', 'b', 'intensity', 'time'])
# Reserved for the control loop, do not use in sequence
Quit = namedtuple('Quit', [])

uris = [
    'radio://0/80/2M/E7E7E7E700',  # cf_id 0
    # Customize URI according to your setup
]

sequence = [
    # Step, CF_id,  action
    (0,    0,      Takeoff(1, 2)),

    (20,    0,      Goto(0.1,  0,   1, 0.75)), #any goto command

    (25,    0,      Goto(0,  0.5,   1, 10)), #second long and overlapping goto
    #drone should drift away towards -x indefinitely

    (40,    0,      Land(2)), #land quickly for safety
]

def activate_high_level_commander(scf):
    scf.cf.param.set_value('commander.enHighLevel', '1')

def crazyflie_control(scf):
    cf = scf.cf
    control = controlQueues[uris.index(cf.link_uri)]

    commander = scf.cf.high_level_commander

    while True:
        command = control.get()
        if type(command) is Quit:
            return
        elif type(command) is Takeoff:
            commander.takeoff(command.height, command.time)
        elif type(command) is Land:
            commander.land(0.0, command.time)
        elif type(command) is Goto:
            commander.go_to(command.x, command.y, command.z, 0, command.time)
            pass
        else:
            print('Warning! unknown command {} for uri {}'.format(command,
                                                                  cf.uri))

def control_thread():
    pointer = 0
    step = 0
    stop = False

    while not stop:
        print('Step {}:'.format(step))
        while sequence[pointer][0] <= step:
            cf_id = sequence[pointer][1]
            command = sequence[pointer][2]

            print(' - Running: {} on {}'.format(command, cf_id))
            controlQueues[cf_id].put(command)
            pointer += 1

            if pointer >= len(sequence):
                print('Reaching the end of the sequence, stopping!')
                stop = True
                break

        step += 1
        time.sleep(STEP_TIME)

    for ctrl in controlQueues:
        ctrl.put(Quit())

if __name__ == '__main__':
    controlQueues = [Queue() for _ in range(len(uris))]

    cflib.crtp.init_drivers()
    factory = CachedCfFactory(rw_cache='./cache')
    with Swarm(uris, factory=factory) as swarm:
        swarm.parallel_safe(activate_high_level_commander)
        swarm.reset_estimators()

        print('Starting sequence!')

        threading.Thread(target=control_thread).start()

        swarm.parallel_safe(crazyflie_control)

        time.sleep(1)
```
</div>

krichardsson, 2022-10-06:
I had a look at this and it indeed seems very strange and I can confirm that is a bug. 

I condensed your example a bit.
It seems as the time in the second go_to changes the speed when it goes away.

``` python
import time
import cflib.crtp
from cflib.crazyflie import Crazyflie
from cflib.crazyflie.syncCrazyflie import SyncCrazyflie
from cflib.utils import uri_helper

uri = uri_helper.uri_from_env(default='radio://0/80/2M/E7E7E7E7E7')

if __name__ == '__main__':
    cflib.crtp.init_drivers()

    with SyncCrazyflie(uri, cf=Crazyflie(rw_cache='./cache')) as scf:
        cf = scf.cf
        cf.param.set_value('commander.enHighLevel', '1')

        commander = cf.high_level_commander

        commander.takeoff(1.0, 2.0)
        time.sleep(2)

        commander.go_to(0.1, 0, 1.0, 0.0, 0.75)
        time.sleep(0.5)

        # The longer time this is, the faster does the CF go in negative X
        long_time = 10.0

        commander.go_to(0, 0.5, 1.0, 0.0, long_time)
        time.sleep(3)

        commander.land(0.0, 2.0)
        time.sleep(2)
        commander.stop()
```

krichardsson, 2022-10-07:
After some more digging this is what I found.

Let's ignore y, z and yaw in the following discussion and focus on x, they are all handled independently.

In the example we first tell the commander to go from x=0 to x=0.1. The initial velocity and acceleration is 0 and the planner will plan a trajectory that also ends with no velocity and acceleration. When we interrupt the first trajectory after 0.5 seconds we should have x: 0.082670, xv: 0.204845, xa: -1.229088, I got x=0.083559, xv=0.189709, x=-1.091556 when testing live. We can see that the drone has almost reached the end point, has some velocity and is retarding to come to a stop.

At this point we tell the commander to re-plan a trajectory to x=0 (again ending with no velocity and acceleration), the planner will use the current position, velocity and acceleration as a starting point and create a trajectory to reach the goal in 10 seconds. The planned trajectory will actually reach 0 after 10 seconds, it just makes a detour to -2.5 on the way (which can be tested in the example above by increasing the sleep to 10 seconds).

To examine this I generated the second polynomial (for my live run) in the firmware using

``` c
float poly[PP_SIZE];
poly7_nojerk(poly, 10.0f, 0.083559f, 0.189709f, -1.091556f, 0.0f, 0.0f, 0.0f)
```
and it generated the polynomial `[0.189709, -0.545778, 0.0, 0.050491, -0.009991, 0.000744, -0.000019]`

If I evaluate it at some selected time points I get:
```
t: 0.00, x: 0.083559, xv: 0.189709, xa: -1.091555
t: 0.25, x: 0.097062, xv: -0.080215, xa: -1.056723
t: 0.50, x: 0.044823, xv: -0.333808, xa: -0.963691
t: 0.75, x: -0.067424, xv: -0.558525, xa: -0.828175
t: 1.00, x: -0.231285, xv: -0.745511, xa: -0.663990
t: 2.0, x: -1.186890, xv: -1.042927, xa: 0.064155
t: 3.0, x: -2.097983, xv: -0.693899, xa: 0.573431
t: 4.0, x: -2.470091, xv: -0.032536, xa: 0.681196
t: 5.0, x: -2.190491, xv: 0.551306, xa: 0.441986
t: 6.0, x: -1.481972, xv: 0.799387, xa: 0.047897
t: 7.0, x: -0.719036, xv: 0.672830, xa: -0.271052
t: 8.0, x: -0.206184, xv: 0.340234, xa: -0.346530
t: 9.0, x: -0.017909, xv: 0.066165, xa: -0.171525
t: 10.0, x: 0.000057, xv: 0.000049, xa: 0.000018
```

We can see that it first continues in the positive direction and then turns back, which makes sense. After this it goes off to -2.5 and turns back to stop at 0. This second part feels a bit odd to me, not sure if it is the intended behavior.

Do you have any input on this @whoenig and @jpreiss ?

krichardsson, 2022-10-07:
After some more digging this is what I found.

Let's ignore y, z and yaw in the following discussion and focus on x, they are all handled independently.

In the example we first tell the commander to go from x=0 to x=0.1. The initial velocity and acceleration is 0 and the planner will plan a trajectory that also ends with no velocity and acceleration. When we interrupt the first trajectory after 0.5 seconds we should have `x: 0.082670, xv: 0.204845, xa: -1.229088`, I got `x=0.083559, xv=0.189709, x=-1.091556` when testing live. We can see that the drone has almost reached the end point, has some velocity and is retarding to come to a stop.

At this point we tell the commander to re-plan a trajectory to x=0 (again ending with no velocity and acceleration), the planner will use the current position, velocity and acceleration as a starting point and create a trajectory to reach the goal in 10 seconds. The planned trajectory will actually reach 0 after 10 seconds, it just makes a detour to -2.5 on the way (which can be tested in the example above by increasing the sleep to 10 seconds).

To examine this I generated the second polynomial (from my live run) in the firmware using

``` c
float poly[PP_SIZE];
poly7_nojerk(poly, 10.0f, 0.083559f, 0.189709f, -1.091556f, 0.0f, 0.0f, 0.0f)
```
and it generated the polynomial [0.189709, -0.545778, 0.0, 0.050491, -0.009991, 0.000744, -0.000019]

If I evaluate it at some selected time points I get:

```
t: 0.00, x: 0.083559, xv: 0.189709, xa: -1.091555
t: 0.25, x: 0.097062, xv: -0.080215, xa: -1.056723
t: 0.50, x: 0.044823, xv: -0.333808, xa: -0.963691
t: 0.75, x: -0.067424, xv: -0.558525, xa: -0.828175
t: 1.00, x: -0.231285, xv: -0.745511, xa: -0.663990
t: 2.0, x: -1.186890, xv: -1.042927, xa: 0.064155
t: 3.0, x: -2.097983, xv: -0.693899, xa: 0.573431
t: 4.0, x: -2.470091, xv: -0.032536, xa: 0.681196
t: 5.0, x: -2.190491, xv: 0.551306, xa: 0.441986
t: 6.0, x: -1.481972, xv: 0.799387, xa: 0.047897
t: 7.0, x: -0.719036, xv: 0.672830, xa: -0.271052
t: 8.0, x: -0.206184, xv: 0.340234, xa: -0.346530
t: 9.0, x: -0.017909, xv: 0.066165, xa: -0.171525
t: 10.0, x: 0.000057, xv: 0.000049, xa: 0.000018
```
We can see that it first continues in the positive direction and then turns back, which makes sense. After this it goes off to -2.5 and turns back to stop at 0. This second part feels a bit odd to me, not sure if it is the intended behavior.

Do you have any input on this [@whoenig](https://github.com/whoenig) and [@jpreiss](https://github.com/jpreiss) ?

jpreiss, 2022-10-07:
As a quick check for numerical issues, you could replace every `float` in ppttraj.h/c with `double`, and use the python bindings to see if anything changes.

krichardsson, 2022-10-07:
I suspected the floats at first, but I don't think that is the problem.
I have re-implemented selected parts of poly7_nojerk() in python to play around with the code a bit faster and get the same behavior

``` python
def polyval(poly, t):
    x = 0.0
    for n in range(8):
        i = 7 - n
        x = x * t + poly[i]
    return x

def poly7_nojerk(T, x0, dx0, ddx0, xf, dxf, ddxf):
    poly = [0.0] * 8

    T2 = T * T
    T3 = T2 * T
    T4 = T3 * T
    T5 = T4 * T
    T6 = T5 * T
    T7 = T6 * T
    poly[0] = x0
    poly[1] = dx0
    poly[2] = ddx0/2
    poly[3] = 0
    poly[4] = -(5*(14*x0 - 14*xf + 8*T*dx0 + 6*T*dxf + 2*T2*ddx0 - T2*ddxf))/(2*T4)
    poly[5] = (84*x0 - 84*xf + 45*T*dx0 + 39*T*dxf + 10*T2*ddx0 - 7*T2*ddxf)/T5
    poly[6] = -(140*x0 - 140*xf + 72*T*dx0 + 68*T*dxf + 15*T2*ddx0 - 13*T2*ddxf)/(2*T6)
    poly[7] = (2*(10*x0 - 10*xf + 5*T*dx0 + 5*T*dxf + T2*ddx0 - T2*ddxf))/T7

    return poly

def print_x(poly, T, steps):
    t = 0.0
    dt = T / steps
    while t <= T:
        pos = polyval(poly, t)
        print(f"t={t}, x={pos}")
        t += dt

def do_test(T, x, xv, xa):
    print(f"Testing T={T}, x={x}, xv={xv}, xa={xa}")
    poly = poly7_nojerk(10.0, 0.083559, 0.189709, -1.091556, 0.0, 0.0, 0.0)
    print_x(poly, T, 10)
    print()

do_test(10.0, 0.083559, 0.189709, -1.091556)
do_test(3.0, 0.083559, 0.189709, -1.091556)
do_test(2.0, 0.083559, 0.189709, -1.091556)
do_test(1.0, 0.083559, 0.189709, -1.091556)
```

jpreiss, 2022-10-07:
That's a good point. It's possible that the 0-jerk boundary condition in the 7th-order solver is somehow antagonistic.

There is now an open-source equivalent of CVXPY: https://github.com/cvxgrp/cvxpygen, if we're willing to settle for a fixed Bézier order and number of pieces. I used CVXPY a long time ago to solve thrust mixing for the tilthex - integrating its output into the Crazyflie firmware was pretty easy.

ataffanel, 2025-02-06:
We will add a warning on overlapping commands.

ArisMorgens, 2025-02-14:
Added warning in [this](https://github.com/bitcraze/crazyflie-lib-python/pull/514) library pull request.

closed 2025-02-14 by ArisMorgens