bitcraze/crazyflie-firmware #1129: P2P library

fbenti, 2022-10-06:
I want to implement some message exchange between crazyflies, and I saw and successfully tested the appchannel_test which exchange message between the copter and the pc throughout the radio. I have used the appchannelTest.py in the tools folder.

However, what I'm interested in is probably the app_peer_to_peer, or the app_p2p_DTR, but there is no such a python file as the channelTest. Can anyone give me some input on how to do this python file (what should be inside, etc..) or redirect me to a guide or anything of the like.

krichardsson, 2022-10-07:
P2P is used to communicate between two Crazyflies. As this is code running on the Crazyflie it is not possible to use python, instead those apps are written in C. 

Is this what you want to do?

fbenti, 2022-10-08:
In the examples folder, there is the app_appchannel_test which is used to exchange messages between the copter and the pc with the radio PA. in the tools subfolder, there's the appchannelTest.py [https://github.com/bitcraze/crazyflie-firmware/tree/master/examples/app_appchannel_test/tools](appchannelTest.py) which, I think, uses the app written in C but I don't understand how. So, running the python script, I can use the app written in C. 

What I would like to do is to create a similar python script but for the P2P app.

krichardsson, 2022-10-10:
The Crazyradio unfortunately don't support P2P so it is not possible to use P2P from a python script.