bitcraze/crazyflie-firmware #1182: Unifying math functions

knmcguire, 2023-01-05:
In both the controllers and estimators, math functions are being rewritten as part of the algorithm, but it would be better to separate those all in a unified math .c files or such. We already have math3d.h and such, but some of the latest contributions has not been as consistent.

Here is a list of those functions that qualify, so more can be added in the future:
- https://github.com/bitcraze/crazyflie-firmware/blob/af96c070d6bc32cf73eeb89980a970d7e01f7f8c/src/modules/src/kalman_core/mm_tdoa_robust.c#L25-L27

whoenig, 2023-01-05:
This should also include the data structures. For example, there are multiple different definitions of 3d vectors and quaternions in the code.

knmcguire, 2023-01-05:
good point! I changed the title.

krichardsson, 2023-01-30:
Also include definitions of pi (I found 3) :-)