bitcraze/crazyflie-firmware #1197: Feature request for additional customization of Lighthouse deck geometry

willgraham345, 2023-01-23:
I'm new to contributing to any contribution on Bitcraze's repo, so forgive me if I make any errors/assumption that are incorrect :)

I was hoping to see if I could get some direction or a additional functionality to edit geometry of the lighthouse position system. I'm part of a research group looking into repositioning the lighthouse positioning deck (https://www.bitcraze.io/products/lighthouse-positioning-deck/) to have the lighthouse position the Crazyflie from beneath, rather than an overhead view. We've begun work on designing a PCB to hold both the Lighthouse deck alongside the Flow deck (https://www.bitcraze.io/products/flow-deck-v2/), to keep the functionality and robustness of the Flow Deck. 

If possible, I'd love to get some guidance on how I should approach this problem and/or work with Bitcraze to create this feature. Thanks!!

ataffanel, 2023-01-24:
Hi,

This is not something we have tested but it should work without many changes. The main change will be to change the [sensor position order](https://github.com/bitcraze/crazyflie-firmware/blob/9e7e697906eced761d5fd27be4f5d866654da250/src/modules/src/lighthouse/lighthouse_position_est.c#L207-L214): the order is used for yaw estimation, if you flip the deck you will flip the sensors position. You will also need to change the position if you offset the sensors, the Z position for example will need to be changed. That way you make sure that the kalman filter is still calculating the position on the same point.

The only thing that we think might cause problems is the yaw estimation the yaw estimation is a bit of a hack looking at the order at which the beam hits the sensors. I think that changing the sensor position order will be enough to make the yaw working but if it is not the case, you can look at the [yaw estimation function](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/utils/src/lighthouse/lighthouse_geometry.c#L169-L201).

In the long run, we should likely make the sensor position a more configurable settings. If you manage to get the deck working on the bottom we can start talking about how to make the change more configurable so that it could be done with a simple build config on the official firmware.

willgraham345, 2023-01-25:
Perfect! Thank you for the guidance, and I'll keep you updated on my work if we're able to get it running.

willgraham345, 2023-05-11:
Hi again! Still working on this problem, and hopefully I'll be able to get it working and contribute to the repository! 

I've been busy working with the STM debugger, and I've noticed that there isn't a way to directly debug key functions related to this (such as the [lighthouse_core](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/lighthouse/lighthouse_core.c)). I've placed a variety of breakpoints within the lighthouse functions and I haven't ever hit them. Is there a way to step through functions such as the [kalman](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/kalman_core/kalman_core.c) using the debugger? 

As of now, I've been running the main.c initialization described within the debugger. If I'd like to try and step through the functions mentioned above should I try to create my own app?

krichardsson, 2023-05-12:
Hi!

Even though your problem with the debugging is related to lighthouse and your work on this issue, I believe it is a general debugging problem that might be interesting for other users as well. Would it be possible for you to ask in our [discussions forum](https://github.com/orgs/bitcraze/discussions) instead where we handle questions like this?

Thanks!

ToveRumar, 2025-06-26:
This seems to be somewhat obsolete and I will close it. Please reopen if needed

closed 2025-06-26 by ToveRumar