bitcraze/crazyflie-firmware #1205: Aideck Wifi (wifi_streamer) not working when using with radio

matthewoots, 2023-01-26:
Hi there I am using crazyswarm2 (https://github.com/IMRCLab/crazyswarm2) and using the wifi streamer and I found a bug, quite a detrimental effect on the streamer performance. When I run with just the `streamer` alone or the `server` alone everything works fine. But once I launch the `server` with the `streamer`, the NINA wifi disconnects **periodically**, there is something amidst and it shows in the printout below.

Do hope to hear from you soon :)

```
[crazyflie_server-4] [INFO] [1674709517.833063892] [cf1]: CPX: ESP32: I (79820) WIFI: Connection accepted
[crazyflie_server-4] [INFO] [1674709517.836556307] [cf1]: CPX: ESP32: I (79820) WIFI: Client connected
[crazyflie_server-4] [INFO] [1674709517.840276241] [cf1]: CPX: CPX connected
[crazyflie_server-4] [INFO] [1674709517.843782529] [cf1]: CPX: GAP8: Wifi client connection status: 1
[crazyflie_server-4] [INFO] [1674709518.096205888] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=48 ms
[crazyflie_server-4] [INFO] [1674709518.276708118] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=113 ms
[crazyflie_server-4] [INFO] [1674709518.439300094] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=114 ms
[crazyflie_server-4] [INFO] [1674709518.626324859] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=92 ms
[crazyflie_server-4] [INFO] [1674709518.799563844] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=124 ms
[crazyflie_server-4] [INFO] [1674709518.982226292] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=107 ms
[crazyflie_server-4] [INFO] [1674709519.187114813] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=120 ms
[crazyflie_server-4] [INFO] [1674709519.360537049] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=136 ms
[crazyflie_server-4] [INFO] [1674709519.427021867] [cf1]: CPX: ESP32: I (81410) WIFI: station:14:85:7f:34:2b:b4leave, AID=1
[crazyflie_server-4] [INFO] [1674709521.870073728] [cf1]: CPX: ESP32: I (83860) WIFI: station:14:85:7f:34:2b:b4 join, AID=1
[crazyflie_server-4] [INFO] [1674709523.719840107] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=113 ms
[crazyflie_server-4] [INFO] [1674709523.889722717] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=4300 ms
[crazyflie_server-4] [INFO] [1674709524.070934341] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=107 ms
[crazyflie_server-4] [INFO] [1674709524.245189427] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=109 ms
[crazyflie_server-4] [INFO] [1674709524.405954740] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=120 ms
[crazyflie_server-4] [INFO] [1674709524.588609782] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=98 ms
[crazyflie_server-4] [INFO] [1674709524.749981732] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=105 ms
[crazyflie_server-4] [INFO] [1674709524.915351907] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=109 ms
[crazyflie_server-4] [INFO] [1674709525.088151712] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=97 ms
[crazyflie_server-4] [INFO] [1674709525.189541017] [cf1]: CPX: ESP32: I (87180) WIFI: station:14:85:7f:34:2b:b4leave, AID=1
[crazyflie_server-4] [INFO] [1674709527.233125606] [cf1]: CPX: ESP32: I (89220) WIFI: station:14:85:7f:34:2b:b4 join, AID=1
[crazyflie_server-4] [INFO] [1674709529.694769547] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=99 ms
[crazyflie_server-4] [INFO] [1674709529.876262988] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=4567 ms
[crazyflie_server-4] [INFO] [1674709530.058607978] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=107 ms
[crazyflie_server-4] [INFO] [1674709530.259809961] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=127 ms
[crazyflie_server-4] [INFO] [1674709530.409766689] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=114 ms
[crazyflie_server-4] [INFO] [1674709530.579326347] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=96 ms
[crazyflie_server-4] [INFO] [1674709530.737578999] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=100 ms
[crazyflie_server-4] [INFO] [1674709530.916279787] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=109 ms
[crazyflie_server-4] [INFO] [1674709530.988861917] [cf1]: CPX: ESP32: I (92980) WIFI: station:14:85:7f:34:2b:b4leave, AID=1
[crazyflie_server-4] [INFO] [1674709533.036425796] [cf1]: CPX: ESP32: I (95020) WIFI: station:14:85:7f:34:2b:b4 join, AID=1
[crazyflie_server-4] [INFO] [1674709535.219833227] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=98 ms
[crazyflie_server-4] [INFO] [1674709535.396508319] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=4251 ms
[crazyflie_server-4] [INFO] [1674709535.559200932] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=109 ms
[crazyflie_server-4] [INFO] [1674709535.730965342] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=103 ms
[crazyflie_server-4] [INFO] [1674709535.889510986] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=103 ms
[crazyflie_server-4] [INFO] [1674709536.063197171] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=96 ms
[crazyflie_server-4] [INFO] [1674709536.250307728] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=104 ms
[crazyflie_server-4] [INFO] [1674709536.431230030] [cf1]: CPX: GAP8: capture=63ms, encoding=0 ms (104976 bytes), transfer=126 ms
[crazyflie_server-4] [INFO] [1674709536.537315509] [cf1]: CPX: ESP32: I (98520) WIFI: station:14:85:7f:34:2b:b4leave, AID=1
[crazyflie_server-4] [INFO] [1674709538.431756784] [cf1]: CPX: ESP32: I (100420) WIFI: station:14:85:7f:34:2b:b4 join, AID=1
[crazyflie_server-4] [INFO] [1674709538.779631218] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=119 ms
[crazyflie_server-4] [INFO] [1674709539.083534342] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=2293 ms
[crazyflie_server-4] [INFO] [1674709539.294160612] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=232 ms
[crazyflie_server-4] [INFO] [1674709539.480803349] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=146 ms
[crazyflie_server-4] [INFO] [1674709539.724258478] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=131 ms
[crazyflie_server-4] [INFO] [1674709539.895213931] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=165 ms
[crazyflie_server-4] [INFO] [1674709540.185958770] [cf1]: CPX: GAP8: capture=62ms, encoding=0 ms (104976 bytes), transfer=112 ms
[crazyflie_server-4] [INFO] [1674709540.324866870] [cf1]: CPX: ESP32: I (102310) WIFI: station:14:85:7f:34:2b:b4leave, AID=1
[crazyflie_server-4] [INFO] [1674709542.145060691] [cf1]: CPX: ESP32: I (104130) WIFI: station:14:85:7f:34:2b:b4 join, AID=1
[crazyflie_server-4] [INFO] [1674709542.336536310] [cf1]: CPX: ESP32: I (104330) WIFI: Client disconnected
[crazyflie_server-4] [INFO] [1674709542.338390224] [cf1]: CPX: CPX disconnected
```

Kaijun101, 2023-01-26:
Hi @matthewoots and @krichardsson,

I am also facing similar issues with streaming images on the AI-Deck and commanding the crazyflie with crazyswarm2. I would like to control the crazyflie swarm while streaming images from the AI-Deck for perception tasks. However, it  always disconnect for no reason.

Really appreciate if anyone can provide any insights to solve this. Currently, stuck at getting reliable wifi streaming for some exciting project with crazyflie hahaha. :)

yanrui89, 2023-01-26:
Hi there, not sure if this is related, but sometimes when I connect the drone to the cfclient after powering up and I look into the console window, it continually publishes that my drone is disconnected from the access point (without me doing any wifi streaming). Is this a cause for concern?

matthewoots, 2023-01-27:
seems like the problem occurs when its full resolution `(324, 324)` while the original `(324,244)` work, is it something to do with the packet size affecting the transfer, since I see a more stable transfer rate here
```
[crazyflie_server-4] [INFO] [1674787288.358091315] [cf1]: CPX: GAP8: capture=59ms, encoding=0 ms (79056 bytes), transfer=64 ms
[crazyflie_server-4] [INFO] [1674787288.491323193] [cf1]: CPX: GAP8: capture=58ms, encoding=0 ms (79056 bytes), transfer=63 ms
[crazyflie_server-4] [INFO] [1674787288.622844429] [cf1]: CPX: GAP8: capture=58ms, encoding=0 ms (79056 bytes), transfer=72 ms
[crazyflie_server-4] [INFO] [1674787288.757375796] [cf1]: CPX: GAP8: capture=58ms, encoding=0 ms (79056 bytes), transfer=69 ms
[crazyflie_server-4] [INFO] [1674787288.886349800] [cf1]: CPX: GAP8: capture=58ms, encoding=0 ms (79056 bytes), transfer=80 ms
```

knmcguire, 2023-01-30:
Thanks for the update! This is indeed interesting. CPX was designed with testing it with the original resolution constantly so that is perhaps a corner case that we missed.

Anyway, @krichardsson has marked this for our next triage, which will be next week when we will discuss this issue.

knmcguire, 2024-09-27:
I've tried it out again on my home network and such with no problem, along with on our office network. I believe that this is more of a network restriction issue. If I run it from a VM within windows I have more issues than running it from natively ubuntu. Since this is not really something we can control or fix at this moment (very esp or network specific) we'll close this issue.

closed 2024-09-27 by knmcguire