bitcraze/crazyflie-firmware #1238: Make bindings_python error 127: Missing dependency in docs.

santiagorg2401, 2023-02-25:
Hi,
I'm currently working on Ubuntu 22 and ROS2 Humble with Crazyswarm2, however, I've been facing an error while trying to build python bindings for Crazyswarm  2, after executing:
```
make cf2_defconfig
make bindings_python
```
I get: 
```
Makefile:219: warning: overriding recipe for target 'bindings_python'
tools/kbuild/Makefile.kbuild:147: warning: ignoring old recipe for target 'bindings_python'
make[1]: Entering directory '/home/santiagorg2401/ros2_ws/crazyflie-firmware/build'
  GEN     ./Makefile
scripts/kconfig/conf  --silentoldconfig Kconfig
make[1]: Nothing to be done for 'bindings_python'.
make[1]: Leaving directory '/home/santiagorg2401/ros2_ws/crazyflie-firmware/build'
swig -python -Isrc/modules/interface -Isrc/hal/interface -Isrc/utils/interface -o build/cffirmware_wrap.c bindings/cffirmware.i
make: swig: No such file or directory
make: *** [Makefile:219: bindings_python] Error 127
```
So I installed SWIG by doing:
`sudo apt -y install swig`

And now it seems to be working.

krichardsson, 2023-02-27:
Thanks for the feedback!
I makes sense that swig is required...

knmcguire, 2023-03-03:
This should be fixed the next time the website is deployed

closed 2023-03-03 by knmcguire