bitcraze/crazyflie-firmware #1244: Kalman flipped resets attitude

knmcguire, 2023-03-03:
With the Kalman filter, if a crazyflie is flipped and the kalman resets, the attitude estimate is reset as well.

[image]

I guess this won't cause issues for the tumble detection as that should be based on sensor data on the accelerometer, but I would imagine this might be very confusing.

Perhaps we should rethink the supervisor resetting kalman a bit? I guess there will be no way to seperate the attitude with the position reset. ?

matejkarasek, 2023-03-13:
Just triggered some LH issue (#1250) causing the EKF to reset in flight. Then this is particularly annoying, as this ends up with a flip and crash...  

I guess the EKF would need to be split somehow such the the attitude part keeps running... Not sure that is possible.

Maybe running complementary filter in parallel and switching to that attitude estimate when position filter needs to be reset?Even better would be to wait till the EKF converges, before switching back... Then we would only get some position drift, but not an immediate crash...

krichardsson, 2023-03-14:
The estimator resets because the state is out of bounds (position far away or velocity very high). A quick test shows that the attitude is equally messed up. 

[image]
The graph is from a Crazyflie on my desk.

I'm not sure it would be possible to only reset the position and velocity, but even if it was the attitude state might not be that useful anyway.