bitcraze/crazyflie-firmware #1258: Attitude rate controller not working

jc-bao, 2023-03-25:
I use the attitude controller in `attitude_pid_controller.c` to control my crazyflie. 
I set the parameter in `crtp_commander_rpyt.c` as zero to use the attitude rate controller. 

```c++
static RPYType stabilizationModeRoll  = RATE; // Current stabilization type of roll (rate or angle)
static RPYType stabilizationModePitch = RATE; // Current stabilization type of pitch (rate or angle)
static RPYType stabilizationModeYaw   = RATE;  // Current stabilization type of yaw (rate or angle)
```

But the attitude rate behaves weirdly. I set a certain roll rate setpoint through crazyswarm2. But the atitude seems goes up a little, then failling back to zero. 

[image]

The dash blue line is my setpoint, while the orange line is the actual drone attitude rate calculated from the attitude. 

I am wondering why the roll rate seem thrust a little bit then falling back to zero quickly? I have converted the command to the [degree measurement ](https://github.com/jc-bao/crazyswarm2-adaptive/blob/e9fc90ab87d88f41f9c5d577d82d2e64da6644af/src/crazyswarm2/crazyflie_examples/crazyflie_examples/attitude_ctl.py#L204) and tried different [PID control parameters](https://github.com/jc-bao/crazyswarm2-adaptive/blob/e9fc90ab87d88f41f9c5d577d82d2e64da6644af/src/crazyswarm2/crazyflie/config/crazyflies.yaml#L18). But things seems not work out. 

Any comments will be highly appreciated. Thanks!

jc-bao, 2023-03-26:
I believe the issue has been resolved. I attempted various controllers and utilized cflib to directly call the attitude rate controller. This led me to discover that the issue was located within my crazyswarm2 settings, specifically regarding the controller type and Kalman filter parameters.

Once I updated the launch file parameters, I was able to achieve a notably improved attitude rate control performance.

[image]

closed 2023-03-26 by jc-bao

gy2256, 2024-02-29:
Hi, whhich flight mode do you choose to enable the attitude rate control?

knmcguire, 2024-03-01:
Could you start a new discussion thread? since you already closed this one

knmcguire, 2024-03-01:
I mean, to start a discussion thread here: https://discussions.bitcraze.io/. We'd like to not have questions in the issue thread unless it constitute something that needs to be fixed.

gy2256, 2024-03-01:
Ok. Thanks!