bitcraze/crazyflie-firmware #1263: Debugger not working with Lighthouse boards

willgraham345, 2023-04-06:
We've been working with the crazyflie debugger (https://www.bitcraze.io/products/debug-adapter-kit/), and have noticed that the debugger doesn't correctly compile/work when we have a lighthouse deck attached. Is this an issue on our end, or is this to be expected with the debugger? If this is an issue on our end, I will provide a more in-depth explanation to try and determine where we went wrong.

knmcguire, 2023-04-11:
Hi! Interesting... and you are saying that debugging works without the lighthouse deck? And are you using GDB directly or openocd-gdb? Perhaps also give us the documentation you started from.

krichardsson, 2023-04-12:
I can unfortunately not reproduce this on my machine. @willgraham345 can you tell us a bit more about what you do?

willgraham345, 2023-04-27:
Sorry for the very late response! 

I think what was happening is that we were only using one Lighthouse at a time. We've realized that we need the LED on the lighthouse board to be green to do any sort of debugging on it. Thank you for your help, and hopefully this issue can be helpful to someone else!