bitcraze/crazyflie-firmware #1283: Trajectories with unaligned offset hard faults

krichardsson, 2023-05-22:
A trajectory can be uploaded and defined with an arbitrary offset in the trajectory memory. When the trajectory is executed, a pointer to the trajectory is cast to [`struct poly4d*`](https://github.com/bitcraze/crazyflie-firmware/blob/c4925370e3c8a35a79bc1e34f9799028751222f6/src/modules/src/crtp_commander_high_level.c#L634). When one of the floats in the struct is accessed, we will get a hard fault If the trajectory is not properly aligned in the trajectory memory. 

One solution could be to change the API to make sure the offset used in the trajectory memory is aligned on 4 byte boundaries, or perhaps it is possible to memcpy data to an aligned working memory during trajectory execution. 

Related to #1282

knmcguire, 2023-10-10:
Is this fixed now with that PR?

knmcguire, 2023-10-12:
Apparently this is still an issue but since nobody has complained so far it has a low-priority to fix.