bitcraze/crazyflie-firmware #1299: atof() function does not compile

krichardsson, 2023-06-19:
Converting strings to floats using the `atof()` function does not work, some libraries seems to be missing.
It has been confirmed to work in 2022.01 and my guess is that it got lost in the introduction of kbuild.

krichardsson, 2023-08-22:
This seems to be caused by the `-nostdlib` flag that was added to LDFLAGS in the Makefile, when converting to KBUILD

whoenig, 2023-08-22:
Did you compare the size of the resulting image? I think not using stdlib can save both flash and potentially improve the runtime due to better optimizations. I am also surprised that there is even a use-case for converting a string to a float number.

krichardsson, 2023-08-23:
Yes, and the results are (drum roll..):

With `-nostdlib`
Flash |  265452/1032192 (26%),  766740 free | text: 257500, data: 7952, ccmdata: 0
RAM   |   82700/131072  (63%),   48372 free | bss: 74748, data: 7952
CCM   |   62024/65536   (95%),    3512 free | ccmbss: 62024, ccmdata: 0

Without `-nostdlib`
Flash |  271188/1032192 (26%),  761004 free | text: 264212, data: 6976, ccmdata: 0
RAM   |   81740/131072  (62%),   49332 free | bss: 74764, data: 6976
CCM   |   62024/65536   (95%),    3512 free | ccmbss: 62024, ccmdata: 0

So the cost is 5-6k of flash and 960 bytes of RAM

> I am also surprised that there is even a use-case for converting a string to a float number.

I could imagine using it in a configuration file, like the SD card config for instance, but I don't think it is that common :-)

closed 2023-08-24 by krichardsson