bitcraze/crazyflie-firmware #1329: Flowdeck based estimate seems to diverge faster

knmcguire, 2023-10-31:
So with the release testing, I found out that flowdeck based flight is a bit iffy.... It seems to diverge faster than usually!

Some things that worked:
* moving the flowdeck higher from the ground (if you have some long header pins)
* Waving the crazyflie to get some flow info before flying.

Still, the general perspective is that it 'used' to work much better than before. There is no black and white error here but it is good to investigate what recent changes caused this.

knmcguire, 2023-11-15:
Users have started to notice this too, especially when it is still on the ground. We are going to look to this a bit more closer in the next couple of weeks https://github.com/orgs/bitcraze/discussions/1033. First step is to figure out which release it still worked okay

williamleong, 2024-03-07:
Hi @knmcguire, we have been using [2022.12] as you suggested and it seems to have improved flow deck performance (otherwise the Crazyflie just shoots off and doesn't stop on some floor types).

However, I am trying to use the `cpxRegisterAppMessageHandler` function to receive messages from the AI deck, which seems to be only introduced in [2023.06]. Is there an alternative? I am wondering how it would have been done before the introduction of `cpxRegisterAppMessageHandler`.

knmcguire, 2024-03-07:
@williamleong 

Your question about cpx, please ask it on https://discussions.bitcraze.io/. This is out of scope for this ticket, which is here to indicate an issue that needs to be fixed on th flowdeck

williamleong, 2024-03-07:
Apologies, I meant to ask if this would be fixed soon so that we could use the latest version with the flow deck as well as the cpx. In the meantime I've manually merged some code to get it to work.

knmcguire, 2024-03-08:
I think the flowdeck based flight still works okay, but there are some circumanstances to keep in mind. I wrote a blogpost about it here https://www.bitcraze.io/2023/11/go-with-the-flow-relative-positioning-with-the-flow-deck/.

CPX question should still be referred to discussions still