bitcraze/crazyflie-firmware #1344: Integrating Custom Out of Tree Controller with Python Bindings

jc-bao, 2024-02-02:
I recently followed the instructions from [this Bitcraze blog post](https://www.bitcraze.io/2023/02/adding-an-estimator-or-controller/) to integrate my custom controller into the firmware, and I found the process quite straightforward and effective. However, I've encountered an issue when attempting to test this custom controller using the Python bindings in Crazyswarm's software-in-the-loop (SIL) simulation. It appears that the out-of-tree (OOT) controller isn't compiled with the Python bindings, and as a result, I'm unable to call it in my tests.

I'm reaching out to the community for assistance on this matter. Does anyone have experience or insights on how to properly call an OOT controller using Python bindings in this context? Alternatively, if there's a recommended workaround or best practice for testing custom controllers before deploying it to the hardware, I would greatly appreciate your guidance.

Thank you in advance for your help and suggestions!

knmcguire, 2024-02-05:
@whoenig do you perhaps have a bit of an opinion on this? Or perhaps some words of advise? Would we want it to be possible to easily include the OOT controller ?

whoenig, 2024-02-05:
In principle this should be very easy, as long as we define how the filename should always be. One needs to update https://github.com/bitcraze/crazyflie-firmware/blob/master/bindings/cffirmware.i and https://github.com/bitcraze/crazyflie-firmware/blob/master/bindings/setup.py. Since it is OOT, we could probably work with the CONFIG_CONTROLLER_OOT variable and conditionally include the OOT source.

Is there an example repo that has a mock OOT controller? I couldn't find a link in the Bitcraze blog post.

knmcguire, 2024-02-05:
Ah no there is not. I believe that the blog post's example only exists in the text there. Do you need the example to exist in the firmware repo then?

whoenig, 2024-02-05:
The strange thing about the blog post is that it's actually not OOT and instead suggests to add a new folder inside the examples of the firmware.
I created a proper OOT example here: https://github.com/IMRCLab/crazyflie-firmware-oot (based on my understanding of https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/development/oot/). It compiles fine and cleanly separates the firmware from the custom controller.

Now the issue is that https://github.com/bitcraze/crazyflie-firmware/blob/master/tools/make/oot.mk doesn't have a target to make the bindings... There also seem many other features missing, e.g., the option to do an OOT build for Bolt, or debugging. I am not entirely sure how to add the bindings without code duplication - perhaps a make or kbuild expert could help out here?

knmcguire, 2024-02-07:
Thanks for the example @whoenig! I was also a bit confused when I read the blogpost. Perhaps we should integrate this into examples instead. 

For kbuild or make.. that would also require a bit of time from our side to look at it... It has been marked for triage.

knmcguire, 2024-03-19:
Result from the triage: So, this is a bit of an edge case and since wolfgang provided an good example so we will close this for now.

closed 2024-03-19 by knmcguire

cameronbed, 2024-07-01:
@knmcguire I am attempting to run an OOT controller out of the examples folder. I believe everything is configured correctly but I am wondering if you know of an existing method to determine the controller at runtime? I am looking through the cfclient while connected but don't see a parameter value for an OOT controller being actively used.

knmcguire, 2024-07-02:
hi @cameronbed, could you start a discussion thread here: https://discussions.bitcraze.io/. This issue has been closed.