bitcraze/crazyflie-firmware #1347: Fail to connect crazyflie in cfclient via radio after flashing to firmware 23.11

xiaoshan-lin, 2024-02-12:
We have been switching between firmware 22.05 and 23.11 several times without an issue. In the latest attempt, we are unable to connect crazyflie in cfclient via radio after flashing to firmware 23.11.

- We are still able to connect to the crazyflie via usb. Here is the output of the console.

        SYS: Crazyflie 2.1 is up and running!
        SYS: Build 0:e527895e8c6e (2023.11) CLEAN
        SYS: I am 0x203336473152500700250024 and I have 1024KB of flash!
        CFGBLK: v1, verification [OK]
        DECK_CORE: 0 deck(s) found
        IMU: BMI088: Using I2C interface.
        IMU: BMI088 Gyro connection [OK].
        IMU: BMI088 Accel connection [OK]
        IMU: BMP388 I2C connection [OK]
        ESTIMATOR: Using Complementary (1) estimator
        CONTROLLER: Using PID (1) controller
        MTR-DRV: Using brushed motor driver
        SYS: About to run tests in system.c.
        SYS: NRF51 version: 2023.11
        EEPROM: I2C connection [OK].
        STORAGE: Storage check [OK].
        IMU: BMI088 gyro self-test [OK]
        SYS: Self test passed!
        STAB: Wait for sensor calibration...
        SYS: Free heap: 20496 bytes
        STAB: Starting stabilizer loop
        SUP: Can not fly
        SUP: Ready to fly

- We cannot turn off the crazyflie by pressing the button.
- M2 and M3 LEDs are solid blue, while M1is blinking red.
- After unplugging the usb, we press "scan" multiple times in the cfclient. It didn't detect any availabe radio. 

Thank you for your suggestions and help.

gemenerik, 2024-02-13:
Connect to the Crazyflie over USB and in cfclient check your config (“Connect->Configure 2.X”). It may have been cleared.

xiaoshan-lin, 2024-02-14:
Thanks for the hint. I attached a [screenshot](https://github.com/bitcraze/crazyflie-firmware/assets/79180119/0d84f28f-31ce-4c3a-a4e1-545cc08dbea1) of the config here. Cfclient still can't find the crazyflie by its radio uri after we press "scan". We can't turn off the crazyflie by pressing the button. 

When we flash the crazyflie, a wireless charging deck is attached to the crazyflie. But it wasn't charging. Could this be the cause of the issue?

gemenerik, 2024-02-15:
Perhaps a bit trivial, but are you scanning for the correct radio address in cfclient? So in the top left of the main cfclient ui is the address set to 0xE7E7E7E708?

The charging deck should not affect this behavior.

xiaoshan-lin, 2024-02-15:
Yes we am indeed scanning for the correct uri. We tried a [recovery firmware flashing](https://www.bitcraze.io/documentation/repository/crazyflie-clients-python/master/userguides/recovery-mode/) and was able to fix this issue. Thank you for your help!

closed 2024-02-15 by xiaoshan-lin