bitcraze/crazyflie-firmware #1360: Failed to modify the default effect of LED-ring deck

LIU-Xueming, 2024-04-02:
Hi, the default effect of LED-ring deck is ID: 6 (double spinner ). Now, i tried to add a new effect, named solidYellowEffect(), and set it to the default effect. But it failed, and I need some help.

1. First, the static function is add in the file _**src/deck/drivers/src/ledring12.c**_

```
static void solidYellowEffect(uint8_t buffer[][3], bool reset)
{
  int i;

  if (reset)
  {
    for (i=0; i<CONFIG_DECK_LEDRING_NBR_LEDS; i++) {
      buffer[i][0] = 255;
      buffer[i][1] = 255;
      buffer[i][2] = 0;
    }
  }
}
```

2. Then, this effect is added in the structure
```
Ledring12Effect effectsFct[] =
{
  blackEffect,
  whiteSpinEffect,
  colorSpinEffect,
  tiltEffect,
  brightnessEffect,
  spinEffect2,
  **_doubleSpinEffect,_**
  solidColorEffect,
  ledTestEffect,
  batteryChargeEffect,
  boatEffect,
  siren,
  gravityLight,
  virtualMemEffect,
  fadeColorEffect,
  rssiEffect,
  locSrvStatus,
  timeMemEffect,
  lighthouseEffect,

  **_solidYellowEffect,_**
};
```

3. Next, I modify the ` src/deck/drivers/src/Kconfig`

```
    config DECK_LEDRING_DEFAULT_EFFECT
    int "Default light effect to use on the LED ring"
    depends on DECK_LEDRING
    default 19
    help
        Index of the default light effect that the LED ring should use
        after boot. Use zero to turn off the LEDs when booting. For more
        information about the LED ring effects see the parameter documentation
        on https://www.bitcraze.io/
```
4. At last, I  compiled and flashed the firmware follow  the document `https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/building-and-flashing/build/`

However, after reboot, the default LED effect is still double spinner, e.g. ID:6.

If, I replace serial number in the structure, e.g.
```
Ledring12Effect effectsFct[] =
{
  blackEffect,
  whiteSpinEffect,
  colorSpinEffect,
  tiltEffect,
  brightnessEffect,
  spinEffect2,

  **_solidYellowEffect_**,

  **_doubleSpinEffect_**,
  solidColorEffect,
  ledTestEffect,
  batteryChargeEffect,
  boatEffect,
  siren,
  gravityLight,
  virtualMemEffect,
  fadeColorEffect,
  rssiEffect,
  locSrvStatus,
  timeMemEffect,
  lighthouseEffect,
};
```
The default LED effect changed as **_solidYellowEffect_**.

Thus, I guess there maybe somewhere in the file I also need to modify the default options, but I didn't find it.
Can you help me see what the problem is? Thanks!

knmcguire, 2024-04-03:
Ah thanks for reporting this! we'll need to find time to recreate this. I hope it is not blocking yo0u too much at the moment?

LIU-Xueming, 2024-04-06:
> Ah thanks for reporting this! we'll need to find time to recreate this. I hope it is not blocking yo0u too much at the moment?

Hi, thanks for your reply.  I recently found an interesting problem.  In fact, I bought crazyflie twice, over half a year apart.

Some crazyflies had the problem above. But others do not have these problems. Maybe this information can help someone.

ToveRumar, 2024-04-10:
Hi!

Interesting issue. This looks indeed like some board discrepancy. 
I have not been able to reproduce the issue myself so far. 

Could you tell me approximately how many of the boards has this problem? Was the boards with this issue purchased at the same time?

ToveRumar, 2024-04-24:
Hi @LIU-Xueming!

Since this looks like a board issue, could you please come back either by email or by starting a new issue if this is a problem for you. 

I'm closing this issue for now since we can not reproduce with the current information.

closed 2024-04-24 by ToveRumar