bitcraze/crazyflie-firmware #1369: Enable motors when crazyflie is upside down.

ZihengQiu, 2024-04-23:
Hello! I want to conduct system identification of crazyflie2, and I place the drone upside down on top of a bottle which is similar to the method mentioned in your official web page. 
But unfortunately, the motor will not run in this upside-down situation, I have attempted to modify some if-conditions in the _crazyflie-firmware/blob/master/src/drivers/src/motors.c_, but still no luck. So is there any way to solve this?

gemenerik, 2024-04-23:
Check out the [documentation on the supervisor](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/functional-areas/supervisor/), too. In `crazyflie-firmware/src/modules/src/supervisor_state_machine.c` there are several transitions based on whether the quadcopter is 'tumbled' or not (`SUPERVISOR_CB_CONF_IS_TUMBLED`).

ZihengQiu, 2024-04-23:
It works, thanks so much!

closed 2024-04-23 by ZihengQiu