bitcraze/crazyflie-firmware #1370: make bindings_python error

Guohao23, 2024-04-23:
ros2 :galactic ubuntu :20.04
guo@Guo:~/crazyflie-firmware$ make bindings_python
Makefile:213: warning: overriding recipe for target 'bindings_python'
tools/kbuild/Makefile.kbuild:147: warning: ignoring old recipe for target 'bindings_python'
make[1]: Entering directory '/home/guo/crazyflie-firmware/build'
make[1]: Nothing to be done for 'bindings_python'.
make[1]: Leaving directory '/home/guo/crazyflie-firmware/build'
swig -python -Isrc/modules/interface -Isrc/hal/interface -Isrc/utils/interface -Isrc/modules/interface/controller -Isrc/platform/interface -Isrc/modules/interface/outlierfilter -Isrc/modules/interface/kalman_core -o build/cffirmware_wrap.c bindings/cffirmware.i
python3 bindings/setup.py build_ext --inplace
In file included from src/modules/interface/kalman_core/kalman_core.h:56,
                 from build/cffirmware_wrap.c:2757:
src/utils/interface/cf_math.h:35:10: fatal error: arm_math.h: No such file or directory
   35 | #include "arm_math.h"
      |          ^~~~~~~~~~~~
compilation terminated.
error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
make: *** [Makefile:214: bindings_python] Error 1

knmcguire, 2024-04-23:
You forgot to do `--recursive` as per instructions. Do `git submodule init` and then `git submodule update`

Duplicate of https://github.com/IMRCLab/crazyswarm2/discussions/486

jpreiss, 2024-05-17:
One situation where you get this error: if you clone a repo whose master branch is super old (like 2+ years), its submodule for CMSIS points to a different repo. If you then check out a feature branch that's more recent, the submodule url won't get updated automatically. A `git submodule sync && git submodule update` will fix it.

knmcguire, 2024-05-17:
@Guohao23 could you confirm that this fixed your problem? then we can close this issue

ataffanel, 2024-11-08:
This issue is stalled so I am closing it. Do not hesitate to re-open if needed.

closed 2024-11-08 by ataffanel