bitcraze/crazyflie-firmware #1376: lee controller weird hop with landing when using lighthouse

knmcguire, 2024-05-16:
With the new lee controller, if the cfclient is used for take off and landing, there is a weird hop at the end that makes the Crazyflie crash. For more information, see the content of this PR: https://github.com/bitcraze/crazyflie-firmware/pull/1343

jglane, 2025-06-27:
Don't know if this was figured out already, but this is because the position integral gains (the vector `Kpos_I`) is set to zero by default, which introduces a steady-state error when controlling with position setpoints. Then when it lands it tries to go "below ground" which creates the hop. I have addressed this in my proposed changes to the Lee controller #1494 but for now, increasing the integral gain parameters (`ctrlLee.Kpos_Ix`, `ctrlLee.Kpos_Iy`, `ctrlLee.Kpos_Iz`) should make it work.

After some testing, I found the hop was gone and the controller behaved quite well, although I was noticing the hopping a few months ago. I noticed in #1465 the default mass for the stock Crazyflie 2.1+ `CF_MASS` was increased from `0.027f` to `0.03454f`. This  made the steady state error nearly unnoticable, but there should still be some steady-state error rejection because different drones have different biases. Also there seems to be little reason behind this change because it doesn't actually weigh that much.

I changed `CF_MASS` back to `0.027f` to test with an "unfavorable" mass setting and set the position integral gains to 3.6 and it seems to work quite well.

ArisMorgens, 2025-07-11:
Hello @jglane!

Testing the controller with an unrealistic mass value is a good point. I was just thinking of setting the default integral gain so that the user doesn't experience the weird hop when selecting the Lee controller without modifying the mass parameter. Do you think that's ok?