bitcraze/crazyflie-firmware #1389: Bolt: one motor doesn't spin

AdrienGuenard, 2024-07-01:
Hello,
I am facing a bug with recent version of firmware built for bolt config: 1 of the 4 motors doesn't spin. Has anyone else noticed the problem? I identified that this problem may come from commit 3d5129ff6afbde9bfe12cc70cb0dfaf948423316

tobbeanton, 2024-07-03:
I can confirm the [problem](https://github.com/bitcraze/crazyflie-firmware/commit/3d5129ff6afbde9bfe12cc70cb0dfaf948423316#diff-f0379ea184fe2cefaa1e47311527c3a7f32d1b863f9c0a3ff682fe6110e74b96R246), it is the Crazyflie brushless ESC reset pin that conflicts with the Bolt M4 mosfet. We will try to have a fix soon.

tobbeanton, 2024-07-03:
A fix #1391 has now been submitted. Please try with that.

AdrienGuenard, 2024-07-03:
Thank you, it seems solved!

closed 2024-07-03 by AdrienGuenard