bitcraze/crazyflie-firmware #1390: "make cload" error in WSL2

Yuhang-Yohoo, 2024-07-02:
Hello devs!
I had some problems when I was trying to upload the firmware via Crazyradio and debug in Vscode, WSL2, Windows.
The instruction I followed is  
[https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/development/openocd_gdb_debugging/](this).
I am not sure if this is caused by WSL or my configuration.

The  first problem happened when I tried to **make cload**:

[image]

The second when I tried to connect to the drone through emulator:
[image]
[image]
[image]

Any suggestions will be appreciated!

Yuhang-Yohoo, 2024-07-03:
I solved both of them. 

## for the first problem
I finally realized it was caused by the Python fault in Windows. To make it work, be sure:
- Upgrade python and pip in **Windows**.
- Check the PATH is set added in **environment variables**(mine is in user variable).
- **Set the target crazyfile to BootLoader mode** before "make cload"

The last step is omitted by the "on-chip debugging" instruction but can be found in other [documents](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/building-and-flashing/build/).   

## for the second problem 

My emulator is Jlink12 from SEEGER, I tried to reach the cortex via Jlink console and found **the Crazyfile chip can only be visited in SWD mode**. So the solution is ：
- in launch.json, change first item of configFiles to "interface/jlink.cfg"
- add a new line in jlink.cfg : "transport select swd". You can find it in openocd/scripts/interface.

And then I can enter the breakpoint in main.c.

Aslo another suggestion is [this tutor](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/development/openocd_gdb_debugging/) includes some deprecated expressions, an updation will be helpful for new beginners.

ataffanel, 2024-07-05:
Thanks a lot for posting the answer! I started looking at the documentation and indeed it is not up to date with the latest WSL development. I have started working on updating it.