bitcraze/crazyflie-firmware #1393: High Level Commander does not run trajectories with ID greater than or equal to 10.

Acoliver102, 2024-07-17:
The documentation states that the CF is capable of calling up to 31 trajectories, but this bug is preventing the drone from using more than 10. From my testing, this does not appear to be an issue with the size of the memory on the CF, seeing as it persists even though the trajectory definition is being sent without any offset and when only one trajectory is being written to the drone.

**Recreation Steps:**
- Use https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/autonomy/autonomous_sequence_high_level.py
- Change trajectory_id in line 159 to any value greater than or equal to 10.
- Run the file.

For arbitrary ID values between 1 and 9, the code behaves as expected. 

_Expected behavior:_ Drone takes off, performs the figure 8 trajectory, and then lands.
_Actual behavior:_ Drone takes off, waits for the duration of the figure 8 trajectory, and then lands.

Tested on:
- cfclient 2024.7.1 / cflib 0.1.26
- cfclient 2023.6 / cflib 0.1.23

CF Firmware version is Firmware: ec6a7929d5b1 (CLEAN). The issue also persists when using Firmware: 182455bcda62 (CLEAN).

Acoliver102, 2024-07-17:
The 2nd Firmware version is forked from the May 2024 version of the CF Firmware, for those curious.

knmcguire, 2024-07-22:
Thanks for notifying us on this! This seems to be a clear bug but I do think that not many upload as many trajectories as you have tried:)

Currently we are a bit low on staff at the office but I've marked this for our next triage for hopefully after the summer. If you happen to find the fix, feel free to make a PR and link to this issue.

Acoliver102, 2024-07-22:
@knmcguire 

Found the issue - constant value that caps trajectory limits was set to 10. Changing it to 31 fixes the issue - system behaves as expected. Can upload and run up to 31 trajectories at once. Pull request is up at https://github.com/bitcraze/crazyflie-firmware/pull/1394.

Thanks for the re

knmcguire, 2024-07-24:
closed with https://github.com/bitcraze/crazyflie-firmware/pull/1394

closed 2024-07-24 by knmcguire