bitcraze/crazyflie-firmware #1405: Crazyflie not booting when one deck is forced and a CRTP controller is used

ataffanel, 2024-09-02:
From bitcraze/discussions#1531: When the Crazyflie is compiled with one forced deck and a CRTP controller is used, the Crazyflie is not starting properly.

First step would be to reproduce this problem in the stab lab, since this is at least the second time we see that. What is likely happening is that there is a flood of CRTP packet reaching the STM32 just at startup and it is somehow locking something. Currently most of the use-case of the Crazyflie, and all of the automated test, have a very quiet radio communication at startup.

Thanks @matejkarasek for reporting the bug!

ataffanel, 2024-09-02:
I think to simulate the problem the test should look something like:
 1. Send hand-crafted commander packet using the python crazyradio driver directly (ie. not using the lib) on the Crazyflie channel/address
 1. Start or restart the Crazyflie under test by power-cycling it
 1. After ~10 seconds, stop sending the commander packets
 1. Try to connect the Crazyflie

@ToveRumar is this something we could already test in the stab lab?

ToveRumar, 2024-09-02:
> @ToveRumar is this something we could already test in the stab lab?

Yes, we definitely could and since this seem to be a bug that reappeared, its a good candidate.

matejkarasek, 2025-11-07:
I have tested this on the flapper platform and it indeed seems #1550 fixes this issue, thanks!