bitcraze/crazyflie-firmware #1421: A serious calculation error was discovered in the lighthouse positioning system

madzhen, 2024-10-15:
For the lighthouse positioning system, taking the first generation of lighthouse as an example, the rotor period of each base station is 8333us, that is, 8.333ms. According to the data from various literatures I have consulted, when the laser scans to the center of the base station, it corresponds to 4000us.

The following data can be used to support this:
[1.https://hacks.esar.org.uk/vive-position-sensor/)
![center](https://github.com/user-attachments/assets/417ba53d-eb3c-4c88-a0f7-90ac04a76d37](https://hacks.esar.org.uk/vive-position-sensor/)
[image]
[2.https://www.researchgate.net/publication/322842913_HIVE_Tracker_A_tiny_low-cost_and_scalable_device_for_sub-millimetric_3D_positioning](https://www.researchgate.net/publication/322842913_HIVE_Tracker_A_tiny_low-cost_and_scalable_device_for_sub-millimetric_3D_positioning)
[image]

According to the above picture, the scanning pulse is in the range of 1222~6777. Then (6778-1222)/2+1222=4000, which just shows that the center position is 4000us

Then the following formula is valid:

`angle = (<time offset in µs> - <central time offset = 4000>) * PI / <cycle period = 8333>`

But the calculation formula I saw in line 205 of pulse_processor_v1.c of crazyfile's firmware is as follows:

```
float frameWidth = stateV1->frameWidth[stateV1->currentBaseStation][stateV1->currentAxis];  
float center = frameWidth/4.0f;  
float angle = (delta - center)*2*(float)M_PI/frameWidth;  
```

I checked carefully and found that the frameWidth here is two rotation cycles, which is about  `8333*2=16666us`. 
**_So the center here is 4166.5, which is a large deviation from 4000,_** which will obviously cause positioning errors.

Can anyone explain why it is calculated like this here? Do you need to correct the center value to 4000?

madzhen, 2024-10-15:
[image]
[image]
This paper also gives the same formular

ataffanel, 2024-10-16:
Hi,

We have not observed huge positioning errors in Lighthouse V1 and this code has been there and used for a while already. Have you observed problems when using Crazyflie with basestation-V1?

This problem will not propagate to lighthouse-V2 since the two systems have completely different frame times, lighthouse V2 using different rotation speed per channels.

madzhen, 2024-10-16:
> Hi,
>
> We have not observed huge positioning errors in Lighthouse V1 and this code has been there and used for a while already. Have you observed problems when using Crazyflie with basestation-V1? [...]

This error does not cause a huge positioning error. This error will cause a deviation between the base station posture identified by the localization method and the actual posture of the base station. For example, if the base station is placed horizontally, if this error exists, it will be identified as tilted.

The new problem I raised will cause deviations:
[https://github.com/bitcraze/crazyflie-firmware/issues/1423](https://github.com/bitcraze/crazyflie-firmware/issues/1423)

That problem will cause the deviation to be greater the farther the positioning board is from the base station. Crazyfile does have this problem at present, and the positioning error at a distance will be about 1cm. I hope you can check whether it is a symbol error.

I also checked the project you referenced, Nasa's astrobee, and your phase correction was transplanted from nana's astrobee project. I carefully derived the mathematical formulas in it and checked the astrobee code. I am sure there is a symbol inconsistency here. The following figure shows the inconsistency between the astrobee code and the crazyfile code:

the following figure is your version：
[image]

the following figure is nasa astrobee version：
[image]

compare the compGib term, astrobee version is different from yours.

closed 2024-10-21 by madzhen