bitcraze/crazyflie-firmware #1424: 1194937331@qq.com

CasperLiZ, 2024-10-16:
I encountered a restart problem. I added a function to record the restart reason in the source code. Sometimes the software restart or pin restart will occur.
[image]
What is the cause? NRF's P0.20 is connected to STM32's NRST pin. Is the cause of the restart controlled by NRF or by cfclient? Thank you very much.
[image]

ataffanel, 2024-10-16:
First of all, this is an interesting functionality, feel free to clean up the code and propose a PR, this could be something useful to have in the firmware.

The nRF51 does hold the reset pin at each Crazyflie start and has the ability to reset the stm32, so yes, either you have some electronic noise triggering the reset (we have never seen that). So the most likely is that the nRF is somehow reseting your stm.

CasperLiZ, 2024-10-17:
I have another question. If we do not consider the impact of nRF51 on the pin reset of stm32, when using CF2.1, the motor sometimes restarts after the self-test sound is connected to cfclient. The "check the reason for the restart" mechanism I wrote will prompt me to "software reset". What is the reason? Or do you have the situation of restarting after connecting to cfclient in normal use?

gemenerik, 2024-10-21:
> If we do not consider the impact of nRF51 on the pin reset of stm32, when using CF2.1, the motor sometimes restarts after the self-test sound is connected to cfclient. The "check the reason for the restart" mechanism I wrote will prompt me to "software reset".

Just to clarify, are you saying that after the Crazyflie completes its self-test (with the startup sound), and you connect to cfclient, the entire system restarts due to a software reset?

CasperLiZ, 2024-10-21:
Yes, thank you very much

gemenerik, 2024-10-23:
That is odd! Are you seeing the same behavior with stock firmware? Has this all been on the same drone? Maybe you can copy the console printout for this problem just like you did in the original post, that would help me debug.