bitcraze/crazyflie-firmware #1430: Propeller test not reporting motorPass

ttolbol, 2024-10-30:
I am trying to get the automatic propeller test to work. I am using python cflib and doing the following:
``` python
cflib.crtp.init_drivers()
scf = SyncCrazyflie(uri, cf=Crazyflie(rw_cache="./cache"))
scf.open_link()
log_config = LogConfig(name="Motor Health", period_in_ms=100)
log_config.add_variable("health.motorPass", "uint8_t")
with SyncLogger(scf, log_config) as logger:
    scf.cf.param.set_value("health.startPropTest", "1")
    for log_entry in logger:
        print(log_entry[1])
```                

I am getting `{'health.motorPass': 0}` despite all 4 propellers being on.
When doing the test, two of the propellers spin, then I get 4 beeps (like when powering on the CF), followed by a repeating pattern of 4 quick red blinks and a pause. 

* How do I know when the test has concluded?
* Why am I not seeing anything in `health.motorPass` after running the test?
* How do I get the drone back in a state where it can fly after the test?

tobbeanton, 2024-11-01:
Perhaps there is something not working when running it from a script. Do you get the same behaviour if you launch the test from the cfclient->console tab?

ttolbol, 2024-11-01:
Interestingly it does not find the lighthouse deck after restarting.

ttolbol, 2024-11-01:
The assertion is here:

```c
void rateSupervisorTask(void *pvParameters) {
  while (1) {
    // Wait for the semaphore to be given by the stabilizerTask
    if (xSemaphoreTake(xRateSupervisorSemaphore, M2T(2000)) == pdTRUE) {
      // Validate the rate
      if (!rateSupervisorValidate(&rateSupervisorContext, xTaskGetTickCount())) {
        if (!rateWarningDisplayed) {
          DEBUG_PRINT("WARNING: stabilizer loop rate is off (%lu)\n", rateSupervisorLatestCount(&rateSupervisorContext));
          rateWarningDisplayed = true;
        }
      }
    } else {
      // Handle the case where the semaphore was not given within the timeout
      DEBUG_PRINT("ERROR: stabilizerTask is blocking\n");
      ASSERT(false); // For safety, assert if the stabilizer task is blocking to ensure motor shutdown
    }
  }
}
```

Any insight on how to fix it is greatly appreciated.

tobbeanton, 2024-11-08:
Fixed in https://github.com/bitcraze/crazyflie-firmware/pull/1432

tobbeanton, 2024-11-08:
I was way to fast. The PR does not fix this issue.

PioBeat, 2025-01-08:
I can confirm the bug with the Crazyflie 2.1+ running Firmware 2024.10.2, when performing the propeller test in the cfclient.
However, the drone still operates normally so far.

For reference, I’ve included the console log below:

```
SYS: Crazyflie 2.1 is up and running!
SYS: Build 0:b046745801fc (2024.10.2) CLEAN
SYS: I am 0x3230383135335112003B001E and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_CORE: 1 deck(s) found
DECK_CORE: Calling INIT on driver bcFlow2 for deck 0
ZR2: Z-down sensor [OK]
PMW: Motion chip id: 0x49:0xB6
IMU: BMI088: Using I2C interface.
IMU: BMI088 Gyro connection [OK].
IMU: BMI088 Accel connection [OK]
IMU: BMP388 I2C connection [OK]
ESTIMATOR: Using Kalman (2) estimator
CONTROLLER: Using PID (1) controller
MTR-DRV: Using brushed motor driver
SYS: About to run tests in system.c.
SYS: NRF51 version: 2024.10 (CF21)
EEPROM: I2C connection [OK].
STORAGE: Storage check [OK].
IMU: BMI088 gyro self-test [OK]
DECK_CORE: Deck 0 test [OK].
SYS: Self test passed!
STAB: Wait for sensor calibration...
SYS: Free heap: 15048 bytes
STAB: Starting stabilizer loop
ESTKALMAN: State out of bounds, resetting
SUP: Can not fly
SUP: Ready to fly
HEALTH: Acc noise floor variance X+Y:0.002429, (Z:0.002471)
HEALTH: Motor M1 variance X+Y: 64.88 (Z:36.45), voltage sag:0.60
HEALTH: Motor M2 variance X+Y: 96.08 (Z:56.51), voltage sag:0.61
SYS: ----------------------------
SYS: Crazyflie 2.1 is up and running!
SYS: Build 0:b046745801fc (2024.10.2) CLEAN
ash!
CFGBLK: v1, verification [OK]
DECK_INFO: Reading deck nr:0 [FAILED]. No driver will be initialized!
DECK_CORE: 0 deck(s) found
IMU: BMI088: Using I2C interface.
IMU: BMI088 Gyro connection [OK].
IMU: BMI088 Accel connection [OK]
IMU: BMP388 I2C connection [OK]
ESTIMATOR: Using Complementary (1) estimator
CONTROLLER: Using PID (1) controller
MTR-DRV: Using brushed motor driver
SYS: About to run tests in system.c.
SYS: NRF51 version: 2024.10 (CF21)
EEPROM: I2C connection [OK].
STORAGE: Storage check [OK].
IMU: BMI088 gyro self-test [OK]
SYS: The system resumed after a failed assert [WARNING]
SYS: Assert failed at ../src/modules/src/stabilizer.c:276
SYS: cfAssertNormalStart [FAIL]
```

ToveRumar, 2025-06-27:
This is fixed in 2025.02. Please try it out and open a new issue if needed.

closed 2025-06-27 by ToveRumar