bitcraze/crazyflie-firmware #1434: Error Cloning Submodules

cookacola, 2024-11-07:
I'm currently using macos with the toolbelt. After cloning recursively, it seems like there are issues cloning the correct submodules, especially CMSIS.

cookacola, 2024-11-07:
Specifically, 
Unable to checkout 'a65b7c9a3e6502127fdb80eb288d8cbdf251a6f4' in submodule path 'vendor/CMSIS'

cookacola, 2024-11-21:
fatal: the remote end hung up unexpectedly
fatal: Unable to checkout 'a65b7c9a3e6502127fdb80eb288d8cbdf251a6f4' in submodule path 'vendor/CMSIS'

Still this error persists.

ataffanel, 2024-11-28:
I cannot reproduce the error on my side. Can you try to clone using ssh instead, this way you might get more quota from GitHub to download cmsis:
```
git clone git@github.com:bitcraze/crazyflie-firmware --recursive
```

cookacola, 2024-11-29:
Now that works. It seems that it's an issue only to HTTPS. Thank you.

cookacola, 2024-11-29:
SSH Works

cookacola, 2024-11-29:
Still receiving this error even with SSH

git-lfs filter-process: git-lfs: command not found
fatal: the remote end hung up unexpectedly
fatal: Unable to checkout 'a65b7c9a3e6502127fdb80eb288d8cbdf251a6f4' in submodule path 'vendor/CMSIS'

cookacola, 2024-11-29:
However, it seems that if I do install git-lfs, it should work.

closed 2024-11-29 by cookacola