bitcraze/crazyflie-firmware #1435: Python bindings bug and update

knmcguire, 2024-11-12:
The spiral and updated go_to PR (https://github.com/bitcraze/crazyflie-firmware/pull/1410) caused compatibility issues with the Crazyswarm2 with the python bindings. 

See this issue: https://github.com/IMRCLab/crazyswarm2/issues/587 

We need to update the plan_to function in the python bindings and add the spiral function itself.

Also let's keep an eye on this PR too: https://github.com/bitcraze/crazyflie-firmware/pull/1377. This will also cause incompatibility with Crazyswarm2... maybe it is better to first merge this and then fix both issues together.

knmcguire, 2024-11-12:
@matejkarasek just letting you know about this. I was half working on this already but haven't gotten the time.

matejkarasek, 2024-11-21:
Thanks for heads up, and my apologies... Unfortunately I have zero experience with Python bindings and by quickly looking into the code I could not find the pieces of code that need to be updated. 
Will keep an eye on this issue to learn and see if I can help resolve it...

knmcguire, 2024-11-21:
No problem. The current plan is to first merge #1377 and fix the bindings in one PR after that.