bitcraze/crazyflie-firmware #1438: Active marker deck and Multi-ranger deck will not work together due to I2C address collision

tobbeanton, 2024-11-28:
Even though the Active marker deck and Multi-ranger deck both are top decks they could still be used together. The compatibility matrix also says so. However when using them together the "MR: Init right sensor [FAIL]"

After investigating it we found that it is due to a I2C address collision. The Active marker deck has address 0x2E and the right multi-ranger sensor gets assigned the same address.

Temporary fix is to manually go in and change the new address space for the multi-ranger [here](https://github.com/bitcraze/crazyflie-firmware/blob/7c64c98060db7622c77c808d97797abb59ef9e9f/src/drivers/src/vl53l1x.c#L52) and add e.g. +10 instead.

lyharvill, 2024-12-10:
This is also true for the lighthouse deck. 
The I2C address of the lighthouse deck is apparently 0x2F. 
The temporary fix for for the I2C address space above fixes the problem. I just tested it on the Bolt.
Might be good to have I2C addresses used by decks on the compatibility matrix.

gemenerik, 2024-12-20:
Adding these to the compatibility matrix may be a good idea.

Are you saying the Lighthouse address also collides with a Multi-ranger address?

lyharvill, 2024-12-20:
Yes,
I tested the Multi-ranger with the Lighthouse deck, and the Lighthouse deck
would not work until I moved the base I2c address of the Mult-ranger deck
in firmware.
Thanks,
Les

lyharvill, 2024-12-20:
Additionally, I physically masked the up TOF sensor on the Multi-ranger
deck since the IR frequency of the two decks are the same.

ataffanel, 2025-02-06:
Can be fixed by changing the start address of the multiranger sensors.

lyharvill, 2025-02-07:
Hi,
Yes, I changed line 52 in src\drivers\vl53I1x.c to...
static int nextI2CAddress = VL53L1X_DEFAULT_ADDRESS+10;  // avoid conflict
with lighthouse deck.
This fixed the conflict.
Thanks,
Les

ToveRumar, 2025-02-10:
Hi @lyharvill !

Thank you so much  for testing it out!
We will fix this issue for the next release of the firmware