bitcraze/crazyflie-firmware #1439: Upgrading Lighthouse Deck firmware, encountered issues

ddwy, 2024-12-02:
When I tried to flash the latest firmware into the lighthouse deck, I was unable to do so and it ended up showing the following：

`Flashing donel Deck update skipped in ColdBoot mode.`

ddwy, 2024-12-02:
When I flash in the latest firmware, the firmware of the Lighthouse slot will be updated together with the Crazyflie firmware when it cannot be flashed from the client.

Bananazooka, 2024-12-03:
Hi, 
Decks are not updated if you use the recovery mode - have you used [those instructions ](https://www.bitcraze.io/documentation/repository/crazyflie-clients-python/master/userguides/recovery-mode/)when flashing? What happens if you follow [those steps](https://www.bitcraze.io/documentation/repository/crazyflie-clients-python/master/installation/install/#installing-from-latest-release) for installing the latest firmware?

ToveRumar, 2025-06-26:
This seems to be a misunderstanding on how to flash the decks. Closing this issue. Please reopen if this is not the case

closed 2025-06-26 by ToveRumar