bitcraze/crazyflie-firmware #1447: uart_syslink assert when flying with APP on some Android versions

ToveRumar, 2025-01-13:
The syslink queue fills up and gets completely full when flying using the APP in some scenarios. 

This needs a bit further investigation but we know that it fills up very quick when flying at least on some Android versions (Samsung S23, android 14).

This causes an assert on `SYS: Assert failed at ../src/drivers/src/uart_syslink.c:549` and the drone just drops from the sky.
This can be mitigated by increasing the queue size or avoiding assert (instead dropping the packet), however we would like to find the actual issue.

zanecodes, 2025-01-14:
After looking through the Android app code a bit more, I decided to try enabling the `Bluetooth latency workaround` setting, and this seems to have fixed the issue completely. I have been able to fly the Crazyflie with the Android app for 10-15 minutes at a time and have not yet experienced the sudden shutdown behavior.

ToveRumar, 2025-01-17:
Hi @zanecodes !

Thank you for looking into this! This really helps us!

I tried this as well, and it seems to have fixed the issue (by what it looks like this is not the first time someone comes across this issue;).

I made sure to document this in the getting started guide. 
Im closing this issue now.

closed 2025-01-17 by ToveRumar