bitcraze/crazyflie-firmware #1449: Threading multiple crazyflies (not using swarm class), erratic flying

john-amadora, 2025-01-22:
I coded in control of 3 synchronous drones via the python wrapper for the crazyflie and using only the motion commander class for movement. I did this instead of the swarm class because each drone will be having its own command and the swarm class is limited to sending setpoints, synchronous movement, and sequential movement. 

 I implemented my system using threading in python. That is, the crazyflie and the motion commander is initialized in a function and the thread is enabled three times in main all with its own URI. 

When one drone is enabled, hovering is steady, but when 3 drones are enabled, the drones tend to drift by a lot. 

How can I fix this? I am thinking that perhaps they are sharing state estimates. Any ideas?

gemenerik, 2025-01-22:
Hi John, do you think this is a firmware issue? The problem might be in your application. If so, let's continue our conversation [in a discussion](https://github.com/orgs/bitcraze/discussions). If you think it's a firmware issue, we can continue here. The state estimates are running on-board, so unless you have specifically implemented some kind of state estimate sharing, this cannot be the cause of the issue. It’s easy to make mistakes in threaded Python applications. Perhaps you can share (snippets of) your code in the discussions post you open.

john-amadora, 2025-01-23:
Hi @gemenerik, i posted a discussion. Thanks

closed 2025-01-23 by gemenerik