bitcraze/crazyflie-firmware #1453: Build fails on macOS M2 with stdio.h errors

ishaanamahajan, 2025-02-04:
Hi,

I was following the official macOS build instructions, the build fails with `stdio.h` related errors in `fixdep.c`. This occurs despite having the correct ARM toolchain installed.

### Environment
- macOS Version: Sonoma 14.6
- gcc version: 11.4.0 (Homebrew)
- Also tried with gcc 14.1.0_2 and 14.2.0

Part of Error Log - 

```
../scripts/basic/fixdep.c:455:17: note: include '<stdio.h>' or provide a declaration of 'fprintf'
make[2]: *** [scripts/basic/fixdep] Error 1
make[1]: *** [scripts_basic] Error 2
make: *** [sub-make] Error 2
```

I will be happy to provide more information if needed, thank you!

ToveRumar, 2025-02-05:
Hi @ishaanamahajan!

Im sorry you are having issues building the FW. Thank you for providing such detailed information about your system. 

What MacBook are you running on, is it an M1? This architecture could cause some issues, from what I have seen.
Did you make sure the submodules are fetched? If you are unsure please run;
`git submodule update --init --recursive` 

How did you install the arm toolchain? 

I just tried installing on a Mac M1 with macos 14.5, and a fresh arm toolchain installed using brew. This worked fine but I needed to install rosetta since im on an M1. However those issues looked different than yours. I installed rosetta like this; 
`softwareupdate --install-rosetta` 

I dont think this is ideal, but for now it could be a workaround. 

Could you try this and see if it helps you?

ishaanamahajan, 2025-02-09:
Hi @ToveRumar thank you for replying! I am using a M2 pro, and I did a brew install for the armchain. 

I tried your workaround but I still get the same errors.

ToveRumar, 2025-02-10:
Hi @ishaanamahajan !

Sorry, I had missed that you wrote the model of the processor in the heading. 
It looks like other people facing similar issues on M2 have been helped by uninstalling their arm gcc toolchain and installing "gcc-arm-embedded" from brew instead.
`brew install --cask gcc-arm-embedded`

Even though it seems counter intuitive since  it should only work for Intel chips according to the brew documentation. However, I do think it is worth a try!

If this does not work I will create an issue and we will have to figure out how this can be solved.

ataffanel, 2025-04-18:
I can confirm, setting up a M4 mac, that ```gcc-arm-embedded``` is required to compile the firmware. I updated the documentation.

commit 10c2997f (2025-04-18): Update Mac build instruction to use gcc-arm-embedded

Fixes #1453
closed 2025-04-18 by ataffanel