bitcraze/crazyflie-firmware #1457: pm.vbat SD Card Logging Does Not Work

jonas-eschmann, 2025-03-05:
Hi, I can't get the `pm.vbat` (or `pm.vbatMV`) to be logged on the SD card. I tried many FW versions (`2023.02`, `2024.10.2`, `2025.02`)

`config.txt`:
```
1     # version
1024  # buffer size in bytes
log   # file name
0     # enable on startup (0/1)
on:fixedFrequency
200     # frequency
1     # mode (0: disabled, 1: synchronous stabilizer, 2: asynchronous)
acc.x
acc.y
acc.z
gyro.x
gyro.y
gyro.z
stateEstimate.qw
stateEstimate.qx
stateEstimate.qy
stateEstimate.qz
rpm.m1
rpm.m2
rpm.m3
rpm.m4
pm.vbat
```

`2024.10.2`:
```
SYS: ----------------------------
SYS: Crazyflie 2.1 is up and running!
SYS: Build 0:b046745801fc (2024.10.2) CLEAN
SYS: I am xxx and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_CORE: 1 deck(s) found
DECK_CORE: Calling INIT on driver bcUSD for deck 0
uSD: mount SD-Card [OK].
IMU: BMI088: Using I2C interface.
uSD: wait for sensors
IMU: BMI088 Gyro connection [OK].
IMU: BMI088 Accel connection [OK]
IMU: BMP388 I2C connection [OK]
ESTIMATOR: Using Complementary (1) estimator
CONTROLLER: Using PID (1) controller
MTR-DRV: Using brushed motor driver
SYS: About to run tests in system.c.
SYS: NRF51 version: 2024.10 (CF21)
EEPROM: I2C connection [OK].
STORAGE: Storage check [OK].
IMU: BMI088 gyro self-test [OK]
DECK_CORE: Deck 0 test [OK].
SYS: Self test passed!
STAB: Wait for sensor calibration...
SYS: Free heap: 16096 bytes
STAB: Starting stabilizer loop
SUP: Can not fly
uSD: Config read [OK].
uSD: malloc buffer 1024 bytes uSD: [OK].
SUP: Ready to fly
uSD: Logging to: log32
uSD: Wrote 72821 B to: log32 (1252 of 1253 events)
```

```
> strings /Volumes/CF/log32 | grep vbat
> strings /Volumes/CF/log32 | grep rpm
rpm.m1(H)
rpm.m2(H)
rpm.m3(H)
rpm.m4(H)
```

Same for `2025.02`:
```
SYS: ----------------------------
SYS: Crazyflie 2.1 is up and running!
SYS: Build 0:16c5630707e9 (2025.02) CLEAN
SYS: I am xxx and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_CORE: 1 deck(s) found
DECK_CORE: Calling INIT on driver bcUSD for deck 0
uSD: mount SD-Card [OK].
IMU: BMI088: Using I2C interface.
uSD: wait for sensors
IMU: BMI088 Gyro connection [OK].
IMU: BMI088 Accel connection [OK]
IMU: BMP388 I2C connection [OK]
ESTIMATOR: Using Complementary (1) estimator
CONTROLLER: Using PID (1) controller
MTR-DRV: Using brushed motor driver
SYS: About to run tests in system.c.
SYS: NRF51 version: 2025.02 (CF21)
EEPROM: I2C connection [OK].
STORAGE: Storage check [OK].
IMU: BMI088 gyro self-test [OK]
DECK_CORE: Deck 0 test [OK].
SYS: Self test passed!
STAB: Wait for sensor calibration...
SYS: Free heap: 16096 bytes
STAB: Starting stabilizer loop
uSD: Config read [OK].
uSD: malloc buffer 1024 bytes uSD: [OK].
uSD: Logging to: log34
uSD: Wrote 43531 B to: log34 (747 of 749 events)
```

Am I doing something wrong, or is this a bug?

whoenig, 2025-03-05:
Make sure to have a new line after "pm.vbat".

jonas-eschmann, 2025-03-05:
Oh good point! It works when only the SD card deck is attached. When the optical flow deck is there as well, it crashes and reboots for me. But just the SD card deck itself is good enough right now. Thank you!

closed 2025-03-05 by jonas-eschmann