bitcraze/crazyflie-firmware #1463: Custom Build for "Kalman Filter Modification for Constant Altitude Flight" Fails to Flash on Crazyflie 2.x

wladimirpetrov, 2025-03-13:
Hello,

I am working with the official firmware release (firmware-cf2-2023.11) that is currently flashed on my Crazyflie 2.x drones. My goal is to eventually modify the firmware’s Kalman filter for constant altitude flight (to help achieve flight above obstacles), using the existing pull request (https://github.com/bitcraze/crazyflie-firmware/pull/823/files#). However, before making any changes, I wanted to verify that I can correctly build the firmware from source and flash it onto the drone.

Here’s what I did:

I cloned the [crazyflie-firmware](https://github.com/bitcraze/crazyflie-firmware) repository.
I configured the build for CF2 with:
`make cf2_defconfig`
I compiled the firmware on Ubuntu 22.04 using:
`make -j$(nproc)`

This produced my custom build (cf2.bin) in the build/ directory.

 I downloaded and unzipped the official firmware release ZIP (firmware-cf2-2023.11), which contains files such as:
`cf2-2023.11.bin
cf2_nrf-2023.11.bin
lighthouse.bin
aideck_esp.bin
manifest.json`

I replaced the official cf2-2023.11.bin with my own custom build by copying my cf2.bin (renamed to match) into the release folder:
`cp ~/crazyflie-firmware/build/cf2.bin ~/Downloads/firmware-cf2-2023.11/cf2-2023.11.bin
`
I then repackaged the folder into a new ZIP:
`cd ~/Downloads/firmware-cf2-2023.11
zip -r my_custom_release.zip *`

I attempted to flash the new ZIP using CFClient by selecting “From file” and choosing my my_custom_release.zip.
However, during the flashing process, the Crazyflie reboots, and the firmware does not appear to be installed. (but if i download the firmware-cf2-2023.11.zip and try to flush it on crazyflies in the same manner it works perfectly)

Questions/Issues:

What am I doing in the wrong way so that I can't flush the build on my cf? Do I have to change the manifest.json?
Is there a specific requirement for packaging a custom build (e.g., specific naming, additional metadata in the manifest) for CFClient to accept and flash the firmware correctly?
Could there be a configuration difference in my build (using cf2_defconfig) versus the official build that results in the bootloader rejecting the firmware? Or I just build it incorrectly?

Any guidance or suggestions to resolve this issue would be greatly appreciated. Thank you for your time!

ArisMorgens, 2025-03-17:
Hello Vladimir!

Thank you for the detailed description of your issue. Your project sounds really fun!
Make sure to follow the [building and flashing](https://github.com/bitcraze/crazyflie-firmware/blob/master/docs/building-and-flashing/build.md) instructions.
The cloning, compiling and building steps that you followed seem to be correct.
Instead of flashing your Crazyflie through the cfclient, I would suggest to automatically enter bootloader mode and flash it by running this command:
`cfloader flash build/cf2.bin stm32-fw -w [CRAZYFLIE_URI]` with `[CRAZYFLIE_URI]` being the uri of the crazyflie.
This is explained [here](https://github.com/bitcraze/crazyflie-firmware/blob/master/docs/building-and-flashing/build.md#automatically-enter-bootloader-mode).

Let us know if this worked or if you need further help on any level of building and flashing the firmware.

wladimirpetrov, 2025-03-17:
Thank you! Your recommendation helped! 

It worked, i was able to flush the firmware on drone. 

Moreover I implemented changes for constant height flight from the pull request, built the firmware and flushed it as well. Now It works perfectly. I am very grateful for your help!