bitcraze/crazyflie-firmware #1466: Help with BH1750 Sensor Integration in Crazyflie Firmware

Nithish190278, 2025-03-24:
Hello Crazyflie Developers,

I am currently trying to integrate a BH1750 light sensor with the Crazyflie firmware. I followed the steps suggested by ChatGPT, including adding a driver file, modifying Makefile, and updating menuconfig. However, I am facing the following issues:

1. BH1750 is not appearing in menuconfig – The option to enable BH1750 does not exist.

2. Build errors – Compilation fails with errors related to i2cdevWrite.

3. The sensor is not detected – Even after making changes, the sensor does not appear in the system logs (console).

Steps I Have Taken:

1. Created src/deck/drivers/src/bh1750.c

#include <stdint.h>
#include <stdbool.h>
#include <stdio.h>
#include "i2cdev.h"
#include "deck.h"

#define BH1750_I2C_ADDRESS 0x23

static bool bh1750Init(DeckInfo *info) {
    uint8_t powerOn = 0x01;

    if (!i2cdevWrite(info->i2cDev, BH1750_I2C_ADDRESS, 1, &powerOn)) {
        printf("BH1750: Initialization failed!\n");
        return false;
    }

    printf("BH1750: Successfully initialized!\n");
    return true;
}

const DeckDriver bh1750Driver = {
    .name = "bh1750",
    .init = bh1750Init,
};

2. Created src/deck/drivers/interface/bh1750.h

#ifndef BH1750_H
#define BH1750_H

#include <stdint.h>
#include <stdbool.h>
#include "deck.h"

bool bh1750Init(DeckInfo *info);

#endif

3. Modified src/deck/deck.c

Added:

extern const DeckDriver bh1750Driver;

Registered bh1750Driver:

const DeckDriver *deckDrivers[] = {
    &bh1750Driver,
};

4. Updated src/deck/drivers/Makefile

obj-y += src/bh1750.o

5. Modified build/configs/config.mk

CONFIG_BH1750=y
CONFIG_DECK_FORCE=y

6. Rebuilt the firmware

make clean
make menuconfig
make -j$(nproc)
make cload

Errors I Am Getting:

Compilation error:

make[3]: *** No rule to make target 'src/deck/bh1750.o', needed by 'src/deck/built-in.o'. Stop.

i2cdevWrite function argument mismatch error:

../src/deck/drivers/src/bh1750.c:19:22: error: passing argument 1 of 'i2cdevWrite' makes pointer from integer without a cast

My Questions:

1. How can I properly add BH1750 support so that it appears in menuconfig?

2. Is there any additional setup required in the Kconfig file to enable the BH1750 option?

3. How do I correctly use i2cdevWrite in Crazyflie firmware?

4. What changes are needed to fix the Makefile and compilation issues?

I would appreciate any guidance, code corrections, or official steps to correctly integrate the BH1750 sensor with Crazyflie firmware.

Thank you!
Best regards,
Nithish Kumar Palla

ataffanel, 2025-04-02:
Hi!

I can first try to answer your question in order and then discuss a bit about deck development:
 1. You can look at the current way decks are added to kbuild: https://github.com/bitcraze/crazyflie-firmware/blob/master/src/deck/drivers/src/Kconfig
 2. If you want your code in the crazyflie firmware, it needs to be compiled in. Adding an entry to menuconfig is one option but it is not actually required.
 3. I suggest you look at other user in the Crazyflie firmware. There are a couple of deck drivers that already use I2C that you can look for reference. For example this seems to look like what you want to do https://github.com/bitcraze/crazyflie-firmware/blob/master/src/deck/drivers/src/acs37800.c
 4. Your files needs to be added to kbuild

Now, I think you should actually follow this documentation and start from there: https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/development/howto/. This tutorial will guide you to the point where you have your code running in the Crazyflie and from there you can implement the code specific to your sensor.

I personally think that starting with chatGPT and going that far with it is not the right way to approach Crazyflie development. It can take you on detours that are hard to come back from. 
 Using documentation and tutorials, like the one I linked above does is a  better approach that will allow you to get things to work step by step.

Nithish190278, 2025-04-02:
Dear Arnaud Taffanel sir,

We attempted to enable the sensor as per your instructions, but we are
still encountering errors. If possible, could you kindly provide any
default code that we can refer to?

We are currently trying to read the sensor values by updating the firmware
using Python. Any guidance or sample code would be greatly appreciated.

Looking forward to your response.

Best regards,
Nithish Kumar

ataffanel, 2025-04-10:
I think the [acs37800.c](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/deck/drivers/src/acs37800.c) driver together with the getting started code can give you a good starting point on how to implement an i2c driver. We unfortunatly do not have other example since I2C driver will be very dependent of what sensor is being used so it is not easy to make a generic example.

closed 2025-04-10 by ataffanel