bitcraze/crazyflie-firmware #1476: Oot estimator not selectable

thelonesomeprogrammer, 2025-04-26:
We are currently trying to implement our own estimator using the built in OutOfTreeEstimator functionality. 

The oot estimater does not seam to be selected in the cfcliet console boot log we have the the oot estimater flag.

After some digging we found that in the stateEstimatorSwitchTo function in the estimator.c file the following section seems to be missing a check for the CONFIG_ESTIMATOR_OOT:
```c
#if defined(CONFIG_ESTIMATOR_KALMAN)
  #define ESTIMATOR StateEstimatorTypeKalman
#elif defined(CONFIG_ESTIMATOR_UKF)
  #define ESTIMATOR StateEstimatorTypeUkf
#elif defined(CONFIG_ESTIMATOR_COMPLEMENTARY)
  #define ESTIMATOR StateEstimatorTypeComplementary
#else
  #define ESTIMATOR StateEstimatorTypeAutoSelect
#endif
```
Any solution, help or commit update would be appreciated.

MesterMaggen, 2025-04-26:
I encountered this code section as well and couldn't see if it was intended or not, but seems to me like the Oot estimator isn't fully accessible yet.

ArisMorgens, 2025-04-28:
Hello @thelonesomeprogrammer and @MesterMaggen!

That's a good catch! I managed to reproduce the issue by selecting the OOT estimator option in the menuconfig.
Then, in the console tab of the cfclient, the OOT estimator is not accessed. However, I did not use any OOT estimator so I can't really validate the results yet.
I pushed some changes on a firmware branch that you can see [here](https://github.com/bitcraze/crazyflie-firmware/compare/master...Aris/OOT_Estimator). 
Let me know if this solved the issue or if we need to further investigate it.

closed 2025-05-09 by ArisMorgens

ArisMorgens, 2025-12-16:
@LumpBloom7 Thanks for pointing that out. The branch has been merged.